isaac-sim/IsaacLab #167: [Question] Does it support external 3d scenes?

user432, 2023-12-10:
Hi, thanks for actively maintaining this huge code base. I would like to know if Orbit supports loading external 3d scenes (like [this](https://3dlg-hcvc.github.io/hssd/)) to train mobile manipulation agents?

Mayankm96, 2023-12-10:
Hi, yes, it should be possible to import these kind of scenes into Orbit. Internally, we have been experimenting with Matterport-3D dataset and have managed to load them into the simulation. The most important part is to generate the USD for those scenes. 

Omniverse supports different asset importers. If you can provide information on what formats the dataset is in, I can provide you pointers to the right extensions to investigate.

user432, 2023-12-10:
Hi Mayank, thanks for your prompt reply. Originally, they're in glb format and I have successfully converted a few of them into USD files using Omniverse's USD composer tool.

Mayankm96, 2023-12-11:
In that case, importing those assets into Orbit should be straightforward. You can use the UsdFileCfg class for this:

> https://github.com/NVIDIA-Omniverse/Orbit/blob/devel/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/spawners/from_files/from_files_cfg.py#L51

Mayankm96, 2024-02-12:
Closing this issue due to inactivity. Please re-open if the issue still persists.

closed 2024-02-12 by Mayankm96

SourabhPrasad, 2024-09-20:
Hi @Mayankm96, I am trying to import the Matterport3D dataset to Isaac Sim 4.1.0. Could you please advice on how you were able to load the dataset?

SourabhPrasad, 2024-10-03:
Thanks for the reply, @Mayankm96!
I have converted the .obj file to .usd, but it doesn't contain the lighting information. Were the lights added manually after conversion to .usd?