isaac-sim/IsaacLab #168: [Question] How to randomize the position of fixed joints?

2361098148, 2023-12-10:
In the reinforcement learning process, there will be some fixed joints whose positions are uncertain, in order to achieve the migration from simulation to real environment, I would like to be able to randomize the positions of the fixed joints. How to get the index of these fixed joints in orbit and randomize their joint positions at reset?

Mayankm96, 2023-12-19:
At the moment, this is not possible using the TensorAPIs of PhysX. You can make this work by going through USD piepline but I don't think it will work either for RL training, since we often disable that for performance reasons.

2361098148, 2023-12-24:
Thanks for your reply! I know it would be hard to implement, but if i can randomize the fixed joint position when the envs are build (or be cloned?). And the joint position will not be changed in the training process. Are there some advices to realize that?

2361098148, 2023-12-24:
Is it possible to add TensorAPIs of PhysX mannually? Can you give some advice on where I should look for help? 

Another solution I envisioned might be this, can I create different usd files and load different usd files when the robots are loaded, for example, if I load 4096 robots normally, can I select 1024 A robots, and the same 1024 each of B\C\D robots

Mayankm96, 2024-04-26:
I think this issue can be resolved through multi-asset spawning as discussed here: https://github.com/NVIDIA-Omniverse/orbit/issues/186

Closing this issue to keep the discussions at the same place.

closed 2024-04-26 by Mayankm96