isaac-sim/IsaacLab #170: [Bug Report] prim view order not correct in FrameTransformer

taochenshh, 2023-12-11:
### Describe the bug

In this [file](https://github.com/NVIDIA-Omniverse/Orbit/blob/devel/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/frame_transformer/frame_transformer.py#L229-L235), it says that the order of elements in `first_env_prim_paths` will be source frame followed by target frames. However, this is not guaranteed. 

In my case, 
`body_names_regex` is `'/World/envs/env_.*/Robot/(fr3_link0|link_3_0_tip|link_15_0_tip|link_7_0_tip|link_11_0_tip)'` where `fr3_link0` is the source frame, but the `first_env_prim_paths` turns out to be:
```
['/World/envs/env_0/Robot/link_3_0_tip',
 '/World/envs/env_0/Robot/link_15_0_tip',
 '/World/envs/env_0/Robot/link_7_0_tip',
 '/World/envs/env_0/Robot/link_11_0_tip',
 '/World/envs/env_0/Robot/fr3_link0']
````

So the source frame is not the first element.

Mayankm96, 2023-12-11:
You are right. The ordering is determined by the PhysX parser and it does not necessarily need to be the order in which the regex expression is defined. We should keep them as two separate views to avoid confusion or re-index the starting frame.

taochenshh, 2023-12-11:
yes, the current code would lead to an error starting from [this line](https://github.com/NVIDIA-Omniverse/Orbit/blob/devel/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/frame_transformer/frame_transformer.py#L235) as this line assumes the order is preseved.

taochenshh, 2023-12-20:
@Mayankm96 will this be fixed soon? One possible way to at least make sure the source frame is not included is replacing [this line](https://github.com/NVIDIA-Omniverse/Orbit/blob/devel/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/frame_transformer/frame_transformer.py#L233) with something like:
```python
target_frame_body_names = [x for x in first_env_body_names  if not re.match(self._tracked_body_names[0], x)]
```

Mayankm96, 2023-12-21:
Hi @taochenshh,

Our hands are tied right now since we are trying to get everything ready to merge devel to main. From the team's availability, we would very likely only be able to take a look after the winter holidays. If you can make a fix, that would be really useful. We can review it and add that patch fix.

Mayankm96, 2024-01-11:
This is now fixed in the commit: https://github.com/NVIDIA-Omniverse/orbit/commit/620ce2b08dc905d0a2b46e986b9da6adbe1f4637

closed 2024-01-11 by Mayankm96