isaac-sim/IsaacLab #171: [Question] Alternative to kit_utils and prim_utils

haastregt, 2023-12-12:
### How can I spawn in floor/lights/objects without using kit_utils or prim_utils?

With Isaac Sim version 2023.1.0, spawning objects with kit_utils and prim_utils results in an error: https://github.com/NVIDIA-Omniverse/Orbit/issues/151. It is suggested to downgrade Isaac Sim, however I need to use Isaac Sim version 2023.1.0 because older versions do not seem to allow sourcing your own ROS workspace https://github.com/NVIDIA-Omniverse/Orbit/issues/164.

I have also attempted to use the devel branch. However it seems that here the kit_utils module as well as some other modules (e.g. the robots module) have been removed. In general it seems that the whole workflow has been changed judging from the demos and tutorials in the devel branch, and it does not seem backwards compatible.

I can imagine that these big changes might be due to the new Isaac Sim version having a different interface. However as I already have simulations using orbit from the main branch that I preferably don't have to rewrite, I was wondering if there is any known workaround in the main branch to spawn in objects without using kit_utils or prim_utils (or using them in a different way) when using Isaac Sim 2023.1.0?

Mayankm96, 2023-12-13:
Hi, I think the error you mentioned in 151 is fairly solvable. We tried summarizing the main API changes in Isaac Sim from 2022.2 to 2023.1 here. Hope it helps!

> https://github.com/NVIDIA-Omniverse/Orbit/blob/devel/docs/source/refs/migration.rst

haastregt, 2023-12-13:
Thank you very much! Exactly what I was looking for :smile:

closed 2023-12-13 by haastregt