isaac-sim/IsaacLab #174: [Question] How to get the positions of Franka Emika's left / right finger?

ninebestwon, 2023-12-18:
### Question

We are conducting a Robot Learning Project using Franka Emika. However, from ```OmniIsaacGymEnvs```, I was able to obtain the Position of Franka Emika's Left/Right Finger with the code below. I would appreciate it if you could tell me how I can obtain the Position of Franka Emika's Left/Right Finger from ```ORBIT```.

```python
from typing import Optional
from omni.isaac.core.articulations import ArticulationView
from omni.isaac.core.prims import RigidPrimView

class FrankaView(ArticulationView):
    def __init__(
        self,
        prim_paths_expr: str,
        name: Optional[str] = "FrankaView",
    ) -> None:

        super().__init__(prim_paths_expr=prim_paths_expr, name=name, reset_xform_properties=False)

        self._hands = RigidPrimView(
            prim_paths_expr="/World/envs/.*/franka/panda_link7", name="hands_view", reset_xform_properties=False
        )
        self._lfingers = RigidPrimView(
            prim_paths_expr="/World/envs/.*/franka/panda_leftfinger", name="lfingers_view", reset_xform_properties=False
        )
        self._rfingers = RigidPrimView(
            prim_paths_expr="/World/envs/.*/franka/panda_rightfinger",
            name="rfingers_view",
            reset_xform_properties=False,
        )
```
```python
    def set_up_scene(self, scene) -> None:
        # first create a single franka
        self.get_franka()

        # call the parent class to clone the single environment
        super().set_up_scene(scene, filter_collisions=False)

        # construct an FrankaView to hold our collection of environments
        self._frankas = FrankaView(
            prim_paths_expr="/World/envs/.*/franka",
            name="franka_view"
        )
        # register the FrankaView object to the world, so that it can be initialized
        scene.add(self._frankas)
        scene.add(self._frankas._hands)
        scene.add(self._frankas._lfingers)
        scene.add(self._frankas._rfingers)

        self.init_data()
        return
```

Mayankm96, 2023-12-19:
Hi, you can obtain all the rigid body states by accessing the corresponding field in the `data` attribute:
> https://github.com/NVIDIA-Omniverse/Orbit/blob/devel/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/rigid_object/rigid_object_data.py#L51-L60

To obtain the indices of the bodies in the tensor, you can use the `find_bodies` function. For performance reasons, I suggest resolving the body indices only once.

> https://github.com/NVIDIA-Omniverse/Orbit/blob/devel/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/rigid_object/rigid_object.py#L143-L155

ninebestwon, 2023-12-22:
I'm glad to hear about the update to ORBIT v0.2.0. However, perhaps due to an update, I can't access the link in your reply. Can you please provide a link to a new version?

Additionally, compared to v0.1.0, the new version of ORBIT appears to be more modularized overall. The project configuration file in "source" has changed a lot. Is it possible to use the environments that were working in the existing v0.1.0 in v0.2.0?

Mayankm96, 2023-12-22:
Hi, you can replace “devel“ with “main“ and the links aabove should work fine again.

Unfortunately, no. We ended up changing so much that the compatibility couldn't remain without having ugly code pathways.

From the migration point of view, if you have your assets and environment working in the previous release, moving to this one should not be that hard. You can compare the locomotion or manipulation environments to see the necessary changes that need to be made.

I apologise for the inconvenience! We will definitely work on keeping things more stable this point onwards.

closed 2024-04-18 by Mayankm96