isaac-sim/IsaacLab #176: [Question] how to increase patch buffer size

taochenshh, 2023-12-19:
Hi, thanks for the great library. I am using Orbit (devel branch) to run 4000 environments in parallel, and I got the following error:

```bash
2023-12-19 20:04:05 [5,983,110ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 219146 in the scene desc!                                  
, FILE /buildAgent/work/74336105e89c4a74/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1240                                                                                                       
2023-12-19 20:04:05 [5,983,409ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 221977 in the scene desc!                                  
, FILE /buildAgent/work/74336105e89c4a74/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1240                                                                                                       
2023-12-19 20:04:05 [5,983,713ms] [Error] [omni.physx.plugin] PhysX error: Patch buffer overflow detected, please increase its size to at least 224549 in the scene desc!                                  
, FILE /buildAgent/work/74336105e89c4a74/source/gpunarrowphase/src/PxgNarrowphaseCore.cpp, LINE 1240
```

Which parameter can I tune to increase this buffer size? Thanks.

Mayankm96, 2023-12-19:
Hi, you can modify the allocated GPU buffers through the setting here:

> https://github.com/NVIDIA-Omniverse/Orbit/blob/devel/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/simulation_cfg.py#L129-L130

closed 2023-12-19 by taochenshh

H-Hisamichi, 2024-09-04:
Hi, @Mayankm96 
I'm also stuck with the same buffer error as @taochenshh . It seems that the URL you provided for the solution doesn't exist.
Could you provide me with the latest information?