isaac-sim/IsaacLab #178: [Bug Report] Bug title

Abi-64, 2023-12-21:
When I run these example:
./orbit.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Reach-Franka-v0 --headless
and a few other examples I get this error:

[INFO]: Completed setting up the environment...
Traceback (most recent call last):
  File "source/standalone/workflows/rsl_rl/train.py", line 126, in <module>
    main()
  File "source/standalone/workflows/rsl_rl/train.py", line 98, in main
    runner = OnPolicyRunner(env, agent_cfg, log_dir=log_dir, device=agent_cfg["device"])
  File "/home/abi/anaconda3/envs/orbit1/lib/python3.7/site-packages/rsl_rl/runners/on_policy_runner.py", line 29, in __init__
    obs, extras = self.env.get_observations()
ValueError: too many values to unpack (expected 2)
Exception ignored in: <function SettingChangeSubscription.__del__ at 0x7f853a9f2a70>
Traceback (most recent call last):
  File "/home/abi/.local/share/ov/pkg/isaac_sim-2022.2.1/Orbit/_isaac_sim/kit/kernel/py/omni/kit/app/_impl/__init__.py", line 114, in __del__
AttributeError: 'NoneType' object has no attribute 'get_settings'
./orbit.sh: line 209: 94352 Segmentation fault      (core dumped) ${python_exe} $@

Any idea why I get this error?

Abi-64, 2023-12-21:
also when I try some libraries for different environments other than the ones in the examples I get the same error.

Mayankm96, 2023-12-21:
Can you please inform the commit, Isaac Sim, and the other details? If you are using `main`, very likely, it is broken right now. I am pretty sure RSL-RL was updated and they changed their APIs for VecEnv.

I highly recommend to use the `devel` branch. We will be merging it soon.

Abi-64, 2023-12-21:
Thanks Mayan, I'm using main, Isaac  2022.2.0. I tried devel branch with isaac 2023.hotfix as well, almost nothing works there. haven't tried devel with 2022.2.0.
Another question, I remember from legged gym we had access to the details of the NN like number of layers and.., I couldn't find the same access on orbit, am I right? 
also Is there any example for sim2real applications?

Abi-64, 2023-12-21:
There is a typo in the installation documents:
# enter the cloned repository
cd orbit
# create a symbolic link
ln -s ${ISAACSIM_PATH} _isaac_sim

orbit should be Orbit, with capital O. 

Also I noticed in devels, the "demo" folder has changed to "demos", so all the paths in the installation guide should change too.

Abi-64, 2023-12-21:
When I install devel branch with Isaac 2022.2.0 I get this error:
ERROR: Ignored the following versions that require a different python version: 0.29.0 Requires-Python >=3.8; 0.29.1 Requires-Python >=3.8; 3.8.0 Requires-Python >=3.8; 3.9.0 Requires-Python >=3.8
ERROR: Could not find a version that satisfies the requirement gymnasium==0.29.0 (from omni-isaac-orbit) (from versions: 0.0.1, 0.26.1, 0.26.2, 0.26.3, 0.27.0, 0.27.1, 0.28.0, 0.28.1)
ERROR: No matching distribution found for gymnasium==0.29.0

Abi-64, 2023-12-22:
I tested 2023.hotfix with devel, this time rsl and rl-gmaes work, SB3 only works for GPU and SKRL doesn't work, I get these errors:
2023-12-22 00:40:10 [4,708ms] [Error] [omni.kit.app._impl] [py stderr]: /home/abi/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/utils/tensorboard/__init__.py:4: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if not hasattr(tensorboard, "__version__") or LooseVersion(

and

2023-12-22 00:40:11 [5,151ms] [Error] [__main__] 'skrl_cfg_entry_point'
2023-12-22 00:40:11 [5,151ms] [Error] [__main__] Traceback (most recent call last):
  File "/home/abi/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/Orbit/source/standalone/workflows/skrl/train.py", line 188, in <module>
    main()
  File "/home/abi/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/Orbit/source/standalone/workflows/skrl/train.py", line 77, in main
    experiment_cfg = load_cfg_from_registry(args_cli.task, "skrl_cfg_entry_point")
  File "/home/abi/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/Orbit/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/utils/parse_cfg.py", line 56, in load_cfg_from_registry
    cfg_entry_point = gym.spec(task_name).kwargs.pop(entry_point_key)
KeyError: 'skrl_cfg_entry_point'

Abi-64, 2023-12-22:
I also noticed in non of the environments the stop button is active 
[image]

Mayankm96, 2024-01-03:
Seems like this has now become a mixture of many reports. While we sincerely appreciate the reporting, we would appreciate it if you make separate issues for each as we can assign priority to them accordingly.

For now, it seems you managed to get the latest main working on 2023.1 release so that issue is resolved. For the others:

- I renamed the repo to be `orbit` now for consistency
- SKRL not working is a known issue. @Toni-SM  for visibility.
- We removed the stop button from standalone execution as it is a non-supported behavior. Earlier when a user pressed the stop button, it is always needed to exit the script. However, this isn't the exact purpose of "STOP" from Omniverse point of view. Thus, we removed it to make it explicit that for closing, you should close the window (and hence the app) itself.

Mayankm96, 2024-01-03:
> Another question, I remember from legged gym we had access to the details of the NN like number of layers and.., I couldn't find the same access on orbit, am I right?

Not sure what you mean here. All training configurations are in the respective environments folder.

> also Is there any example for sim2real applications?

We have done sim-to-real for ANYmal-C locomotion policy. For others environments/tasks, we have not yet investigated them. However, if you find ways to make sim-to-real work reliably, we are open to reviewing such contributions and/or collaborations.

Abi-64, 2024-01-09:
Thanks for the answers Mayan.
Regarding the stop button, if it helps to close the window or training session it would be helpful otherwise we have have to terminate it from terminal. The other thing is that in the headless mode it's almost impossible to stop it from terminal unless you kill the terminal which is a headache, I terminate the terminal 20 times/hour sometimes to test different training parameters.  In the normal mode it would be helpful if we could stop the rending by pression a key like Isaac Gym, I couldn't find this feature in Orbit, right?

Mayankm96, 2024-01-10:
We didn't see the issue that the script did not exit when you do `Ctrl+C` but maybe this depends on the system?

Can you specify your system details and also the script you are running. We definitely should exit the script without a bad experience.

Mayankm96, 2024-02-12:
Closing this issue due to inactivity. We will soon be dropping support for Isaac Sim 2022.2. We encourage all users to move to 2023.1 release since it has many critical fixes that are needed.

closed 2024-02-12 by Mayankm96

Vassil17, 2024-03-14:
I've also encountered this issue under 2023.1.1

### Describe the bug

Cannot exit or stop early any of the training scripts with keyboard interrupts (ctrl+c) when executing from terminal. Happens both when running as ./orbit.sh -p and with python inside the conda env. No error message is returned and the training continues. 

Interestingly, ctrl+c does seem to work when executing scripts in debug mode within vscode. 

Ctrl+Z does exit the script but it leaves the process running (which is annoying as it requires a kill command). It also happens when running scripts from within the docker container.

### Steps to reproduce

Running a training script and pressing CTRL+C.

```
################################################################################
                       Learning iteration 1/1500 

                       Computation: 25420 steps/s (collection: 3.552s, learning 0.315s)
               Value function loss: 0.0094
                    Surrogate loss: -0.0064
             Mean action noise std: 0.98
                       Mean reward: -0.62
               Mean episode length: 35.25
Episode Reward/track_lin_vel_xy_exp: 0.0069
Episode Reward/track_ang_vel_z_exp: 0.0055
       Episode Reward/lin_vel_z_l2: -0.0103
      Episode Reward/ang_vel_xy_l2: -0.0072
     Episode Reward/dof_torques_l2: -0.0047
         Episode Reward/dof_acc_l2: -0.0085
     Episode Reward/action_rate_l2: -0.0086
      Episode Reward/feet_air_time: -0.0010
 Episode Reward/undesired_contacts: -0.0027
Episode Reward/flat_orientation_l2: 0.0000
     Episode Reward/dof_pos_limits: 0.0000
         Curriculum/terrain_levels: 3.4648
Metrics/base_velocity/error_vel_xy: 0.0599
Metrics/base_velocity/error_vel_yaw: 0.0568
      Episode Termination/time_out: 3.4167
  Episode Termination/base_contact: 0.3333
--------------------------------------------------------------------------------
                   Total timesteps: 196608
                    Iteration time: 3.87s
                        Total time: 9.55s
                               ETA: 7156.2s

^C################################################################################
                       Learning iteration 2/1500 

                       Computation: 23830 steps/s (collection: 3.795s, learning 0.330s)
               Value function loss: 0.0110
                    Surrogate loss: -0.0064
             Mean action noise std: 0.97
                       Mean reward: -1.42
               Mean episode length: 64.77
Episode Reward/track_lin_vel_xy_exp: 0.0112
Episode Reward/track_ang_vel_z_exp: 0.0092
       Episode Reward/lin_vel_z_l2: -0.0176
      Episode Reward/ang_vel_xy_l2: -0.0136
     Episode Reward/dof_torques_l2: -0.0079
         Episode Reward/dof_acc_l2: -0.0146
     Episode Reward/action_rate_l2: -0.0143
      Episode Reward/feet_air_time: -0.0015
 Episode Reward/undesired_contacts: -0.0135
Episode Reward/flat_orientation_l2: 0.0000
     Episode Reward/dof_pos_limits: 0.0000
         Curriculum/terrain_levels: 3.4352
Metrics/base_velocity/error_vel_xy: 0.1044
Metrics/base_velocity/error_vel_yaw: 0.0975
      Episode Termination/time_out: 4.0000
  Episode Termination/base_contact: 2.4583
--------------------------------------------------------------------------------
                   Total timesteps: 294912
                    Iteration time: 4.13s
                        Total time: 13.67s
                               ETA: 6827.4s

^C^C^C^C^C^C^C
^C################################################################################
                       Learning iteration 3/1500 

                       Computation: 23569 steps/s (collection: 3.857s, learning 0.314s)
               Value function loss: 0.0113
                    Surrogate loss: -0.0068
             Mean action noise std: 0.96
                       Mean reward: -1.84
               Mean episode length: 89.30
Episode Reward/track_lin_vel_xy_exp: 0.0187
Episode Reward/track_ang_vel_z_exp: 0.0124
       Episode Reward/lin_vel_z_l2: -0.0226
      Episode Reward/ang_vel_xy_l2: -0.0205
     Episode Reward/dof_torques_l2: -0.0110
         Episode Reward/dof_acc_l2: -0.0185
     Episode Reward/action_rate_l2: -0.0196
      Episode Reward/feet_air_time: -0.0018
 Episode Reward/undesired_contacts: -0.0254
Episode Reward/flat_orientation_l2: 0.0000
     Episode Reward/dof_pos_limits: 0.0000
         Curriculum/terrain_levels: 3.3926
Metrics/base_velocity/error_vel_xy: 0.1353
Metrics/base_velocity/error_vel_yaw: 0.1391
      Episode Termination/time_out: 3.7917
  Episode Termination/base_contact: 4.3333
--------------------------------------------------------------------------------
                   Total timesteps: 393216
                    Iteration time: 4.17s
                        Total time: 17.84s
                               ETA: 6678.0s

^Z
[24]+  Stopped                 ./orbit.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Velocity-Rough-Anymal-D-v0 --headless
```
### System Info

Describe the characteristic of your environment:

- Commit: 93f3eff
- Isaac Sim Version: 2023.1.1-rc.8+2023.1.688.573e0291.tc(orbit)
- OS: Ubuntu 20.04
- GPU: RTX 3060
- CUDA: 12.2
- GPU Driver:535.161.07

Mayankm96, 2024-03-14:
@Vassil17 

This commit should have fixed the issue: b95d6f3fb4d98c90e37a0a1439bd52fe3f78c596 

Can you update to latest main and check?

Vassil17, 2024-03-14:
Perfect, I can confirm updating to latest main fixed the issue.