isaac-sim/IsaacLab #179: [Proposal] (Viewer) Camera Manager

btx0424, 2023-12-21:
### Proposal

It would be useful to have control over the default camera (`"/OmniverseKit_Persp"`) for video recording during RL training, e.g., following a quadruped to traverse rough terrains. A simple solution would be adding a function that computes `eye` and `lookat` each time `render` is called.

A rough sketch of its implementation:

```python
@configclass
class ViewerCfg:
    """Configuration of the scene viewport camera."""
    ...
    func: Callable = None
    """

class RLTaskEnv(BaseEnv):
    ...
    def render(self):
        if self.cfg.viewer.func is not None:
            eye, lookat = self.cfg.viewer.func(self)
            self.sim.set_camera_view(eye, lookat)
        ...
```

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

Mayankm96, 2023-12-21:
I had implemented something like this for the UI version:

> https://github.com/NVIDIA-Omniverse/Orbit/blob/devel/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/ui/base_env_window.py#L156-L165

We need to think of a smart way not to repeat this (and other UI-related code) twice. Let's discuss this in January when I'm back from vacation :)

Mayankm96, 2024-02-27:
We implemented a first version of this idea in the commit: 59e6c680aedf0a84be574141258046188c7f5e05

It doesn't have a viewer function itself but does a fixed transform with the operation you mentioned above. If this doesn't complete your use case we can expose the operation as a function too.

Please feel free to open this issue or create a new one.

closed 2024-02-27 by Mayankm96