isaac-sim/IsaacLab #182: [Question] actuator error

Abi-64, 2023-12-24:
Hey guy, thank you for publishing the new version, it's amazing work.
I'm wondering if you have examples for using actuators, I'm trying to use DCMotor actuator for a simple 2 joints robot, in my asset config file I have defined the robot like this:
actuators2={
    "theta1_actuator": DCMotorCfg(
        joint_names_expr= ["theta1"] ,
        # effort_limit=2.0,
        # # velocity_limit=2.0,
        # stiffness=1.0,
        # damping=0.0,
    ),
    "theta2_actuator": DCMotorCfg(
        joint_names_expr= ["theta2"] , 
        # effort_limit=1.0, 
        # # velocity_limit=2, 
        # stiffness=1.0, 
        # damping=1.0
    ),
}

ROBOT1_CFG = ArticulationCfg(
    spawn=spawn1,
    init_state=init_state1,
    actuators=actuators2
)
 
I'm trying to run the joints in different modes in the main code which is a copy of run_articulation.  when the actuator model is ImplicitActuatorCfg it works correctly, but when i change it to DCMotorCfg it generate an error! when i change it to IdealPDActuatorCfg it moves randomly!   could you please help understand what's going on, there is no example for using actuators in the documents,thanks. 

from omni.isaac.orbit_assets import ROBOT1_CFG  # isort:skip

def design_scene() -> tuple[dict, list[list[float]]]:
    """Designs the scene."""
    # Ground-plane
    cfg = sim_utils.GroundPlaneCfg()
    cfg.func("/World/defaultGroundPlane", cfg)
    # Lights
    cfg = sim_utils.DomeLightCfg(intensity=3000.0, color=(0.75, 0.75, 0.75))
    cfg.func("/World/Light", cfg)

    # Create separate groups called "Origin1", "Origin2", "Origin3"
    # Each group will have a robot in it
    origins = [[0.0, 0.0, 0.0], [-1.0, 0.0, 0.0]]
    # Origin 1
    prim_utils.create_prim("/World/Origin1", "Xform", translation=origins[0])
    # # Origin 2
    # prim_utils.create_prim("/World/Origin2", "Xform", translation=origins[1])

    # Articulation
    robot1_cfg = ROBOT1_CFG.copy()
    robot1_cfg.prim_path = "/World/Origin.*/Robot"
    robot1 = Articulation(cfg=robot1_cfg)

    # return the scene information
    scene_entities = {"robot1": robot1}
    return scene_entities, origins

def run_simulator(sim: sim_utils.SimulationContext, entities: dict[str, Articulation], origins: torch.Tensor):
    """Runs the simulation loop."""
    # Extract scene entities
    # note: we only do this here for readability. In general, it is better to access the entities directly from
    #   the dictionary. This dictionary is replaced by the InteractiveScene class in the next tutorial.
    robot = entities["robot1"]
    # Define simulation stepping
    sim_dt = sim.get_physics_dt()
    count = 0
    # Simulation loop
    while simulation_app.is_running():

        jpos=robot.data.joint_pos
        # print(jpos)
        robot.set_joint_position_target(torch.tensor([[.1, .1]]))
        robot.write_data_to_sim()
        # Perform step
        sim.step()
        # Increment counter
        count += 1
        # Update buffers
        robot.update(sim_dt)

def main():

Abi-64, 2023-12-25:
I noticed that DCMotor actutors should be defined like this not like the one in the example, the thing is that if I don't add the armature term it wont settle to the target value and it will vibrate. why?

Motor1_CFG1 = DCMotorCfg(
    joint_names_expr=["theta1"],
    saturation_effort=1,
    effort_limit=1,
    velocity_limit=50,
    stiffness={"theta1": 100.0},
    damping={"theta1": 10.0},
    armature={"theta1": 0.10} 
)

Mayankm96, 2024-01-03:
Well yes. You need to provide these values for the DCMotor to configure them correctly. Otherwise, they are MISSING (error) or None (loads the default values from the USD).

If it is vibrating, then it could very likely be your PD controller is not tuned properly. Adding an armature is similar to having more damping to the joint model.

Also please note that, when using an explicit model, you may require to reduce the simulation timestep to deal with time-step discretization error.