isaac-sim/IsaacLab #184: [Question] Cannot run some of the tutorials and demos in version 1.0

jimmyfu0729, 2023-12-31:
### Question

I have just updated the orbit to the latest version. However, some of the standalone demos and tutorials cannot exert. and the logs are listed below:

**First is for bipeds.py:**
2023-12-31 05:17:59 [6,173ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/ecoMode/enabled has been requested but is not available
2023-12-31 05:17:59 [6,173ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/directLighting/enabled has been requested but is not available
2023-12-31 05:17:59 [6,173ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/reflections/enabled has been requested but is not available
2023-12-31 05:17:59 [6,173ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/translucency/enabled has been requested but is not available
2023-12-31 05:17:59 [6,173ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/raytracing/subsurface/enabled has been requested but is not available
2023-12-31 05:17:59 [6,173ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/caustics/enabled has been requested but is not available
2023-12-31 05:18:00 [7,459ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2023-12-31 05:18:00 [7,459ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[8.419s] app ready
2023-12-31 05:18:02 [8,798ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2023-12-31 05:18:02 [8,798ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[8.994s] RTX ready
[9.317s] Simulation App Startup Complete
[10.624s] [ext: omni.physx.fabric-105.1.12-5.1] startup
2023-12-31 05:18:05 [12,362ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /World/Cassie/Robot_1/pelvis/visuals.proto_mesh_0_id0 that has not been populated
[INFO]: Setup complete...
>>>>>>>> Reset!
Program aborted due to an unhandled Error:
Unable to find target for this triple (no targets are registered)Fatal Python error: Aborted

Thread 0x00007fbf303ff640 (most recent call first):
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.1/kit/python/lib/python3.10/concurrent/futures/thread.py", line 81 in _worker
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.1/kit/python/lib/python3.10/threading.py", line 953 in run
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.1/kit/python/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.1/kit/python/lib/python3.10/threading.py", line 973 in _bootstrap

Current thread 0x00007fc3e1926b80 (most recent call first):
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/rigid_object/rigid_object.py", line 379 in _update_common_data
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py", line 177 in update
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/demos/bipeds.py", line 104 in main
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/demos/bipeds.py", line 110 in <module>

Extension modules: psutil._psutil_linux, psutil._psutil_posix, pydantic.typing, pydantic.errors, pydantic.version, pydantic.utils, pydantic.color, pydantic.networks, pydantic.types, pydantic.error_wrappers, pydantic.config, pydantic.parse, pydantic.main, pydantic.decorator, pydantic.tools, numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, omni.mdl.pymdlsdk._pymdlsdk, torch._C, torch._C._fft, torch._C._linalg, torch._C._nested, torch._C._nn, torch._C._sparse, torch._C._special, scipy._lib._ccallback_c, scipy.sparse._sparsetools, _csparsetools, scipy.sparse._csparsetools, scipy.sparse.linalg._isolve._iterative, scipy.linalg._fblas, scipy.linalg._flapack, scipy.linalg._cythonized_array_utils, scipy.linalg._flinalg, scipy.linalg._solve_toeplitz, scipy.linalg._matfuncs_sqrtm_triu, scipy.linalg.cython_lapack, scipy.linalg.cython_blas, scipy.linalg._matfuncs_expm, scipy.linalg._decomp_update, scipy.sparse.linalg._dsolve._superlu, scipy.sparse.linalg._eigen.arpack._arpack, scipy.sparse.csgraph._tools, scipy.sparse.csgraph._shortest_path, scipy.sparse.csgraph._traversal, scipy.sparse.csgraph._min_spanning_tree, scipy.sparse.csgraph._flow, scipy.sparse.csgraph._matching, scipy.sparse.csgraph._reordering, scipy.spatial._ckdtree, scipy._lib.messagestream, scipy.spatial._qhull, scipy.spatial._voronoi, scipy.spatial._distance_wrap, scipy.spatial._hausdorff, scipy.special._ufuncs_cxx, scipy.special._ufuncs, scipy.special._specfun, scipy.special._comb, scipy.special._ellip_harm_2, scipy.spatial.transform._rotation, PIL._imaging, PIL._imagingft, numpy.linalg.lapack_lite, scipy.ndimage._nd_image, _ni_label, scipy.ndimage._ni_label, osqp._osqp, multidict._multidict, yarl._quoting_c, aiohttp._helpers, aiohttp._http_writer, aiohttp._http_parser, aiohttp._websocket, cchardet._cchardet, _cffi_backend, frozenlist._frozenlist, setproctitle._setproctitle, yaml._yaml, scipy.io.matlab._mio_utils, scipy.io.matlab._streams, scipy.io.matlab._mio5_utils (total: 93)

**And for create_scene.py:**
2023-12-31 05:20:27 [6,164ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/ecoMode/enabled has been requested but is not available
2023-12-31 05:20:27 [6,164ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/directLighting/enabled has been requested but is not available
2023-12-31 05:20:27 [6,165ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/reflections/enabled has been requested but is not available
2023-12-31 05:20:27 [6,165ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/translucency/enabled has been requested but is not available
2023-12-31 05:20:27 [6,165ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/raytracing/subsurface/enabled has been requested but is not available
2023-12-31 05:20:27 [6,165ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/caustics/enabled has been requested but is not available
2023-12-31 05:20:29 [7,484ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2023-12-31 05:20:29 [7,484ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[8.456s] app ready
2023-12-31 05:20:30 [8,833ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2023-12-31 05:20:30 [8,833ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[9.029s] RTX ready
[9.351s] Simulation App Startup Complete
[11.239s] [ext: omni.physx.fabric-105.1.12-5.1] startup
2023-12-31 05:20:33 [12,117ms] [Warning] [omni.physx.tensors.plugin] Failed to find articulation at '/World/envs/env_0/Robot'
2023-12-31 05:20:33 [12,117ms] [Error] [omni.physx.tensors.plugin] Pattern '/World/envs/env_*/Robot' did not match any articulations

2023-12-31 05:20:33 [12,117ms] [Error] [carb.events.python] AttributeError: 'NoneType' object has no attribute 'shared_metatype'

At:
  /home/jimmy/Omniverse/Library/isaac_sim-2023.1.1/extsPhysics/omni.physics.tensors-105.1.12-5.1/omni/physics/tensors/impl/api.py(174): shared_metatype
[... cut ...]
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/tutorials/02_scene/create_scene.py", line 135, in <module>
    main()
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/tutorials/02_scene/create_scene.py", line 129, in main
    run_simulator(sim, scene)
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/tutorials/02_scene/create_scene.py", line 89, in run_simulator
    root_state = robot.data.default_root_state.clone()
AttributeError: 'NoneType' object has no attribute 'clone'

Mayankm96, 2024-01-03:
Please provide more information on the system.

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: [e.g. Ubuntu 20.04]
- GPU: [e.g. RTX 2060 Super]
- CUDA: [e.g. 11.4]
- GPU Driver: [e.g. 470.82.01, this can be seen by using `nvidia-smi` command.]

jimmyfu0729, 2024-01-03:
OS Ubuntu 22.04.3
Isaac Sim 2023.1.1
GPU RTX 3070Ti Laptop GPU

Wed Jan  3 23:36:04 2024       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.147.05   Driver Version: 525.147.05   CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   54C    P0    N/A / 125W |      6MiB /  8192MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      2370      G   /usr/lib/xorg/Xorg                  4MiB |
+-----------------------------------------------------------------------------+

Mayankm96, 2024-01-08:
@jimmyfu0729 Do you have active internet connection? It could be the assets are not being downloaded from the server.

Can you also please share the log file? The log file location should be mentioned in the first three lines or so after you run the script. Something like:

```
Loading user config located at: '/media/vulcan/packman-repo/chk/kit-sdk/105.1+release.130815.04872a84.tc.linux-x86_64.release/data/Kit/Isaac-Sim/2023.1/user.config.json'
[Info] [carb] Logging to file: /media/vulcan/packman-repo/chk/kit-sdk/105.1+release.130815.04872a84.tc.linux-x86_64.release/logs/Kit/Isaac-Sim/2023.1/kit_20240108_113008.log
```

jimmyfu0729, 2024-01-08:
I found some here but don't know if these are relating to the internet problem.

Active user not found. Using default user [kiosk]2024-01-08 14:19:22 [6,513ms] [Error] [carb.livestream.plugin] Stream Server: starting the server failed, 0x800B1002
2024-01-08 14:19:22 [6,513ms] [Error] [carb.livestream.plugin] Could not initialize streaming components
2024-01-08 14:19:22 [6,514ms] [Error] [carb.livestream.plugin] Couldn't initialize the capture device.

and when I want to run add_sensor_on_robot.py, another problem occured:

2024-01-08 14:19:23 [7,932ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/ecoMode/enabled has been requested but is not available
2024-01-08 14:19:23 [7,932ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/directLighting/enabled has been requested but is not available
2024-01-08 14:19:23 [7,932ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/reflections/enabled has been requested but is not available
2024-01-08 14:19:23 [7,932ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/translucency/enabled has been requested but is not available
2024-01-08 14:19:23 [7,933ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/raytracing/subsurface/enabled has been requested but is not available
2024-01-08 14:19:23 [7,933ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/caustics/enabled has been requested but is not available
2024-01-08 14:19:25 [9,605ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2024-01-08 14:19:25 [9,605ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[10.645s] app ready
2024-01-08 14:19:26 [11,024ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2024-01-08 14:19:26 [11,024ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[11.237s] RTX ready
[11.549s] Simulation App Startup Complete
[13.869s] [ext: omni.physx.fabric-105.1.12-5.1] startup
2024-01-08 14:19:33 [18,183ms] [Warning] [omni.fabric.plugin] No source has valid data array=0xa9638bd0 usdValid=0 cpuValid=0
~*~*~*
~*~*~* Direct GPU Helper:
~*~*~*   2 articulations, maxLinks=17
~*~*~*
2024-01-08 14:19:35 [19,541ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /World/envs/env_0/Robot/base/visuals.proto_mesh_21_id21 that has not been populated
2024-01-08 14:19:35 [19,542ms] [Warning] [omni.hydra] Mesh '/World/envs/env_0/Robot/base/visuals.proto_mesh_0_id0' has corrupted data in primvar 'st': buffer size 702 doesn't match expected size 12828 in faceVarying primvars
[INFO]: Setup complete...
Program aborted due to an unhandled Error:
Unable to find target for this triple (no targets are registered)Fatal Python error: Aborted

Thread 0x00007f51f93ff640 (most recent call first):
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.1/kit/python/lib/python3.10/concurrent/futures/thread.py", line 81 in _worker
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.1/kit/python/lib/python3.10/threading.py", line 953 in run
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.1/kit/python/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.1/kit/python/lib/python3.10/threading.py", line 973 in _bootstrap

Current thread 0x00007f56a7978b80 (most recent call first):
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/camera/utils.py", line 337 in convert_orientation_convention
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/camera/camera.py", line 514 in _update_poses
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/camera/camera.py", line 321 in reset
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/scene/interactive_scene.py", line 245 in reset
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/tutorials/04_sensors/add_sensors_on_robot.py", line 135 in run_simulator
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/tutorials/04_sensors/add_sensors_on_robot.py", line 181 in main
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/tutorials/04_sensors/add_sensors_on_robot.py", line 187 in <module>

jimmyfu0729, 2024-01-08:
And this problem happens if the previous problem didn't happen. When I use the old version orbit it worked well.

jimmyfu0729, 2024-01-10:
> We have mostly been testing with Isaac Sim 2023.1.0-hotfix.1 release. Haven't gotten around 2023.1.1 yet, but plan to l look into it this month. If possible, can you try with the 2023.1.0-hotifx.1 please?

I have changed it to the 2023.1.0, but the problem still exist when running the bipeds.py but successfully run the arms.py:

Program aborted due to an unhandled Error:
Unable to find target for this triple (no targets are registered)Fatal Python error: Aborted

Thread 0x00007f220ebff640 (most recent call first):
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/concurrent/futures/thread.py", line 81 in _worker
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/threading.py", line 953 in run
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.0-hotfix.1/kit/python/lib/python3.10/threading.py", line 973 in _bootstrap

Current thread 0x00007f26bc3a1b80 (most recent call first):
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/rigid_object/rigid_object.py", line 379 in _update_common_data
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py", line 177 in update
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/demos/bipeds.py", line 104 in main
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/demos/bipeds.py", line 110 in <module>

Extension modules: psutil._psutil_linux, psutil._psutil_posix, pydantic.typing, pydantic.errors, pydantic.version, pydantic.utils, pydantic.color, pydantic.networks, pydantic.types, pydantic.error_wrappers, pydantic.config, pydantic.parse, pydantic.main, pydantic.decorator, pydantic.tools, omni.mdl.pymdlsdk._pymdlsdk, numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, torch._C, torch._C._fft, torch._C._linalg, torch._C._nested, torch._C._nn, torch._C._sparse, torch._C._special, scipy._lib._ccallback_c, scipy.sparse._sparsetools, _csparsetools, scipy.sparse._csparsetools, scipy.sparse.linalg._isolve._iterative, scipy.linalg._fblas, scipy.linalg._flapack, scipy.linalg._cythonized_array_utils, scipy.linalg._flinalg, scipy.linalg._solve_toeplitz, scipy.linalg._matfuncs_sqrtm_triu, scipy.linalg.cython_lapack, scipy.linalg.cython_blas, scipy.linalg._matfuncs_expm, scipy.linalg._decomp_update, scipy.sparse.linalg._dsolve._superlu, scipy.sparse.linalg._eigen.arpack._arpack, scipy.sparse.csgraph._tools, scipy.sparse.csgraph._shortest_path, scipy.sparse.csgraph._traversal, scipy.sparse.csgraph._min_spanning_tree, scipy.sparse.csgraph._flow, scipy.sparse.csgraph._matching, scipy.sparse.csgraph._reordering, scipy.spatial._ckdtree, scipy._lib.messagestream, scipy.spatial._qhull, scipy.spatial._voronoi, scipy.spatial._distance_wrap, scipy.spatial._hausdorff, scipy.special._ufuncs_cxx, scipy.special._ufuncs, scipy.special._specfun, scipy.special._comb, scipy.special._ellip_harm_2, scipy.spatial.transform._rotation, PIL._imaging, PIL._imagingft, numpy.linalg.lapack_lite, scipy.ndimage._nd_image, _ni_label, scipy.ndimage._ni_label, osqp._osqp, multidict._multidict, yarl._quoting_c, aiohttp._helpers, aiohttp._http_writer, aiohttp._http_parser, aiohttp._websocket, cchardet._cchardet, _cffi_backend, frozenlist._frozenlist, setproctitle._setproctitle, yaml._yaml, scipy.io.matlab._mio_utils, scipy.io.matlab._streams, scipy.io.matlab._mio5_utils (total: 93)

mohamedamr13, 2024-02-05:
> We have mostly been testing with Isaac Sim 2023.1.0-hotfix.1 release. Haven't gotten around 2023.1.1 yet, but plan to l look into it this month. If possible, can you try with the 2023.1.0-hotifx.1 please?

Hi Mayank, 
I am having an error in installing the conda env needed for running orbit on Isaac Sim 2023.1.1 but I also reproduced the same error on 2023.1.0-hotfix. Can you please take a [look](https://github.com/NVIDIA-Omniverse/orbit/issues/218) ?

mohamedamr13, 2024-02-05:
@Mayankm96 

> Hi Mayank, I am having an error in installing the conda env needed for running orbit on Isaac Sim 2023.1.1 but I also reproduced the same error on 2023.1.0-hotfix. Can you please take a [...]

Mayankm96, 2024-02-12:
A shot in the dark, can you uninstall `psutil` and see if that solves the issue?

> ./orbit.sh -p -m pip uninstall psutil robomimic

jimmyfu0729, 2024-02-21:
@Mayankm96 
I found that this problem also happened in the version under orbit 1.0 as I am using the laptop with a intel graphic 770 and a 3070ti laptop. If I input SimulationContext(backend = "torch", device ="cuda") this problem will be solved and everything looks good. But this method cannot work properly in orbit1.0, I don't know why. 

|---------------------------------------------------------------------------------------------|
| Driver Version: 525.147.05    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3070 Ti Lap.. | Yes: 0 |     | 8192    MB | 10de      | 0          |
|     |                                  |        |     |            | 24a0      | 40402025.. |
|     |                                  |        |     |            | 1         |            |
|---------------------------------------------------------------------------------------------|
| 1   | Intel(R) UHD Graphics 770 (ADL.. |        |     | 23841   MB | 8086      | 0          |
|     |                                  |        |     |            | 4680      | 86808046.. |
|     |                                  |        |     |            | 0         |            |
|=============================================================================================|
| OS: 22.04.4 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.4, Kernel: 6.5.0-18-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: 12th Gen Intel(R) Core(TM) i9-12900K | Cores: 16 | Logical: 24
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31789 | Free Memory: 22675
| Total Page/Swap (MB): 0 | Free Page/Swap: 0

[6.352s] Simulation App Starting
2024-02-21 12:48:33 [6,364ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/ecoMode/enabled has been requested but is not available
2024-02-21 12:48:33 [6,364ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/directLighting/enabled has been requested but is not available
2024-02-21 12:48:33 [6,364ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/reflections/enabled has been requested but is not available
2024-02-21 12:48:33 [6,365ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/translucency/enabled has been requested but is not available
2024-02-21 12:48:33 [6,365ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/raytracing/subsurface/enabled has been requested but is not available
2024-02-21 12:48:33 [6,365ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/caustics/enabled has been requested but is not available
2024-02-21 12:48:34 [7,577ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2024-02-21 12:48:34 [7,577ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[8.613s] app ready
2024-02-21 12:48:36 [9,009ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2024-02-21 12:48:36 [9,009ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[9.209s] RTX ready
[9.536s] Simulation App Startup Complete
[11.721s] [ext: omni.physx.fabric-105.1.12-5.1] startup
2024-02-21 12:48:43 [16,883ms] [Warning] [omni.fabric.plugin] No source has valid data array=0xa97281b0 usdValid=0 cpuValid=0
~*~*~*
~*~*~* Direct GPU Helper:
~*~*~*   2 articulations, maxLinks=17
~*~*~*
2024-02-21 12:48:45 [18,118ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /World/envs/env_0/Robot/RH_THIGH/visuals.proto_mesh_1_id1 that has not been populated
2024-02-21 12:48:45 [18,120ms] [Warning] [omni.hydra] Mesh '/World/envs/env_0/Robot/base/visuals.proto_mesh_0_id0' has corrupted data in primvar 'st': buffer size 702 doesn't match expected size 12828 in faceVarying primvars
[INFO]: Setup complete...
Program aborted due to an unhandled Error:
Unable to find target for this triple (no targets are registered)Fatal Python error: Aborted

Thread 0x00007f443ebff640 (most recent call first):
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.1/kit/python/lib/python3.10/concurrent/futures/thread.py", line 81 in _worker
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.1/kit/python/lib/python3.10/threading.py", line 953 in run
  File "/home/jimmy/Omniverse/Library/isaac_sim-2023.1.1/kit/python/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
[... cut ...]
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/tutorials/04_sensors/add_sensors_on_robot.py", line 135 in run_simulator
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/tutorials/04_sensors/add_sensors_on_robot.py", line 181 in main
  File "/home/jimmy/Omniverse/Project_1/Orbit_1.0/Orbit/source/standalone/tutorials/04_sensors/add_sensors_on_robot.py", line 187 in <module>

These are some of the logs, and this problem happens after installed the orbit 1.0 without any manipulation and I just ran the tutorials. Is there any progress in solving this problem?

Mayankm96, 2024-07-01:
This issue seems to have gone outdated. Usually, the issues arise if you have more than one graphic card installed in your system. This includes an Intel card and an NVIDIA card. You need to disable the Intel one to ensure the application uses the right graphic card for Omniverse.

Please reopen this issue if the problem re-arises with the latest IsaacLab.

closed 2024-07-01 by Mayankm96