isaac-sim/IsaacLab #192: [Bug Report] destination /storage doesn't exist in container

LarsDoorenbos, 2024-01-08:
### Describe the bug

Following [these](https://github.com/NVIDIA-Omniverse/orbit/blob/963f3045367be3ccdac4f0902ac2103fcbb6070f/docs/source/deployment/cluster.rst#L20) steps to deploy Orbit on a slurm cluster, there is a fatal error in creating the container stating that the destination /storage doesn't exist in container.

### Steps to reproduce

Follow the steps in the [Cluster guide](https://github.com/NVIDIA-Omniverse/orbit/blob/963f3045367be3ccdac4f0902ac2103fcbb6070f/docs/source/deployment/cluster.rst#L20) to run Orbit on a HPC cluster with slurm.

My local machine starts a job on the cluster, which in turn starts the job that builds the apptainer. The container creation fails due to an error with mounting. The log and error from the job that builds the apptainer looks as follows:

```
(run_singularity.py): Called on compute node with arguments
WARNING: nv files may not be bound with --writable
WARNING: By using --writable, Apptainer can't create /storage destination automatically without overlay or underlay
FATAL:   container creation failed: mount hook function failure: mount /var/apptainer/mnt/session/storage->/storage error: while mounting /var/apptainer/mnt/session/storage: destination /storage doesn't exist in container
(run_singularity.py): Return
```

### System Info

Describe the characteristic of your environment:

- Commit: 963f304
- Isaac Sim Version: 2023.1.0-hotfix.1
- OS (cluster): CentOS Linux 7 (Core)
- GPU: RTX 3090
- CUDA: 12.3
- GPU Driver: 545.23.08

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

masoudmoghani, 2024-01-08:
@AutonomousHansen would you be able to help with this? Thanks

AutonomousHansen, 2024-01-08:
@pascal-roth Do you have any ideas here? The only place I can see that a bind mount potentially causing problems is [here](https://github.com/isaac-orbit/orbit/blob/963f3045367be3ccdac4f0902ac2103fcbb6070f/docker/cluster/run_singularity.sh#L70), but the error message is saying that the directory doesn't exist on the container instead of the host?

pascal-roth, 2024-01-09:
@AutonomousHansen I agree that line is the most probable cause. @LarsDoorenbos, can you ensure that the `logs` directory exists within your orbit directory on the cluster, it won't be synced to it and can be missing the first time you want to run the code.

pascal-roth, 2024-01-09:
that is clear, `/workspace` is defined within your docker image and cannot be accessed from outside. It is the directory where during the docker build process orbit is copied and installed (see [here](https://github.com/isaac-orbit/orbit/blob/963f3045367be3ccdac4f0902ac2103fcbb6070f/docker/Dockerfile#L29))

Try to comment out the line where logs are bound to the image, then we can be certain if this is causing the issue.

LarsDoorenbos, 2024-01-09:
Removing the `-B $CLUSTER_ORBIT_DIR/logs:/workspace/orbit/logs:rw \` line still gives the same error.

pascal-roth, 2024-01-10:
which apptainer version are you using on the cluster?

pascal-roth, 2024-01-11:
I agree; it seems like a general mounting error. It is difficult to reproduce from our side as we are running `apptainer version 1.2.5-1.el7`.

LarsDoorenbos, 2024-01-12:
For now, we found a different machine on which to run Orbit. Thanks anyway!

closed 2024-09-09 by pascal-roth