isaac-sim/IsaacLab #197: [Bug Report] Unable to use RL Games on Isaac-Lift-Cube-Franka-v0

StoneT2000, 2024-01-10:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Trying to run RL games training code one the lift cube task but I get a error here and am not sure what to do fix it.

### Steps to reproduce

Expected 

```
./orbit.sh -p source/standalone/workflows/rl_games/train.py --task Isaac-Lift-Cube-Franka-v0 --headless
```

To run RL on the Isaac Lift Cube Franka-v0 task. However I get the following

```
[3.343s] [ext: omni.isaac.ui-0.13.5] startup
2024-01-10 21:24:00 [3,400ms] [Error] [omni.kit.app._impl] [py stderr]: /home/stao/.local/share/ov/pkg/isaac_sim-2023.1.1/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/utils/tensorboard/__init__.py:4: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if not hasattr(tensorboard, "__version__") or LooseVersion(

/home/stao/.local/share/ov/pkg/isaac_sim-2023.1.1/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/utils/tensorboard/__init__.py:4: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if not hasattr(tensorboard, "__version__") or LooseVersion(
[INFO]: Parsing configuration from: <class 'omni.isaac.orbit_tasks.manipulation.lift.config.franka.joint_pos_env_cfg.FrankaCubeLiftEnvCfg'>
2024-01-10 21:24:00 [4,001ms] [Error] [__main__] 'rl_games_cfg_entry_point'
2024-01-10 21:24:00 [4,001ms] [Error] [__main__] Traceback (most recent call last):
  File "/home/stao/work/external/Orbit/source/standalone/workflows/rl_games/train.py", line 141, in <module>
    main()
  File "/home/stao/work/external/Orbit/source/standalone/workflows/rl_games/train.py", line 71, in main
    agent_cfg = load_cfg_from_registry(args_cli.task, "rl_games_cfg_entry_point")
  File "/home/stao/work/external/Orbit/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/utils/parse_cfg.py", line 56, in load_cfg_from_registry
    cfg_entry_point = gym.spec(task_name).kwargs.pop(entry_point_key)
KeyError: 'rl_games_cfg_entry_point'
```

Full logs here:
```
[INFO] Using python from: /home/stao/miniforge3/envs/orbit/bin/python                                                                                                       
[INFO][AppLauncher]: Loading experience file: /home/stao/work/external/Orbit/_isaac_sim/apps/omni.isaac.sim.python.gym.headless.kit .
Starting kit application with the following args:  ['/home/stao/work/external/Orbit/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/stao/work/external/Orbit/_isaac_sim/apps/omni.isaac.sim.python.gym.headless.kit', '--/app/tokens/exe-path=/home/stao/work/external/Orbit/_isaac_sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/stao/work/external/Orbit/_isaac_sim/exts', '--ext-folder', '/home/stao/work/external/Orbit/_isaac_sim/apps', '--/physics/cudaDevice=0', '--portable', '--no-window']
Passing the following args to the base kit application:  ['--task', 'Isaac-Lift-Cube-Franka-v0', '--headless']
Loading user config located at: '/home/stao/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/data/Kit/Isaac-Sim/2023.1/user.config.json'
[Info] [carb] Logging to file: /home/stao/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/logs/Kit/Isaac-Sim/2023.1/kit_20240110_132356.log
2024-01-10 21:23:56 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2024-01-10 21:23:56 [4ms] [Warning] [omni.ext.plugin] [ext: omni.kit.sequencer.core-103.4.0+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2024-01-10 21:23:56 [4ms] [Warning] [omni.ext.plugin] [ext: omni.kit.sequencer.usd-103.4.0+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2024-01-10 21:23:56 [5ms] [Warning] [omni.ext.plugin] [ext: omni.kit.widget.timeline-105.0.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2024-01-10 21:23:56 [5ms] [Warning] [omni.ext.plugin] [ext: omni.kit.window.sequencer-103.4.2-dev.3+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2024-01-10 21:23:56 [5ms] [Warning] [omni.ext.plugin] [ext: omni.paint.brush.attributes-1.3.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2024-01-10 21:23:56 [6ms] [Warning] [omni.ext.plugin] [ext: omni.usd.schema.sequence-2.3.0+105.0.lx64.r.cp310] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
[0.027s] [ext: omni.kit.async_engine-0.0.0] startup
[0.154s] [ext: omni.assets.plugins-0.0.0] startup
[0.155s] [ext: omni.stats-0.0.0] startup
[0.156s] [ext: omni.client-1.0.1] startup
[0.160s] [ext: omni.gpu_foundation-0.0.0] startup
[0.165s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.166s] [ext: carb.windowing.plugins-1.0.0] startup
[0.171s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.129.03    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24564   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | 2b0351f8.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
[... 156 lines cut ...]
[INFO]: Parsing configuration from: <class 'omni.isaac.orbit_tasks.manipulation.lift.config.franka.joint_pos_env_cfg.FrankaCubeLiftEnvCfg'>
2024-01-10 21:24:00 [4,001ms] [Error] [__main__] 'rl_games_cfg_entry_point'
2024-01-10 21:24:00 [4,001ms] [Error] [__main__] Traceback (most recent call last):
  File "/home/stao/work/external/Orbit/source/standalone/workflows/rl_games/train.py", line 141, in <module>
    main()
  File "/home/stao/work/external/Orbit/source/standalone/workflows/rl_games/train.py", line 71, in main
    agent_cfg = load_cfg_from_registry(args_cli.task, "rl_games_cfg_entry_point")
  File "/home/stao/work/external/Orbit/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/utils/parse_cfg.py", line 56, in load_cfg_from_registry
    cfg_entry_point = gym.spec(task_name).kwargs.pop(entry_point_key)
KeyError: 'rl_games_cfg_entry_point'

[4.026s] Simulation App Shutting Down
2024-01-10 21:24:00 [4,023ms] [Warning] [omni.core.ITypeFactory] Module /home/stao/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
```

### System Info

Describe the characteristic of your environment:

- Commit: 57e766cf68c942191a74e24269b780ee9a817535
- Isaac Sim Version: 2023.1.1-rc.8+2023.1.688.573e0291.tc(orbit)
- OS: Ubuntu 22
- GPU: RTX 4090
- CUDA: 12.2
- GPU Driver: 535.129.03

ozhanozen, 2024-02-11:
This error is due to the fact that the configuration entry point for the  rl_games is indeed missing when the environment is being registered to the gym. 

To add it, you can insert the keyword argument `"rl_games_cfg_entry_point": f"{agents.__name__}:rl_games_ppo_cfg.yaml",` somewhere between lines 22-25 here:

https://github.com/NVIDIA-Omniverse/orbit/blob/57e766cf68c942191a74e24269b780ee9a817535/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/manipulation/lift/config/franka/__init__.py#L18-L28

You need to do the same for all the other environment variations on the same file. This solved the issue for me.

Mayankm96, 2024-02-14:
Thanks @ozhanozen for the answer!

Yes, in general, we provide the facility to train environments with different frameworks. However, it is intractable for us to tune all the frameworks for all the environments we provide. The PPO implementations vary, parameter names are not the same -- all of this makes it not so straightforward.

If you manage to tune them nicely, we are more than happy to welcome MRs with the right settings.