isaac-sim/IsaacLab #203: Pattern '/World/envs/env_*/Robot' did not match any articulations when num_env = 1

zoctipus, 2024-01-14:
This bug is noticed first in my own robot, then also observed in cartpole tutorial in standalone/tutorials/create_scene.py.
Interestingly this is not a issue in franka envs.

To reproduce the bug, simply replacing 

```
scene_cfg = CartpoleSceneCfg(num_envs=args_cli.num_envs, env_spacing=2.0)
```
with 
```
scene_cfg = CartpoleSceneCfg(num_envs=1, env_spacing=2.0)
```

2024-01-14 11:43:17 [17,413ms] [Warning] [omni.physx.tensors.plugin] Failed to find articulation at '/World/envs/env_0/Robot'
2024-01-14 11:43:17 [17,413ms] [Error] [omni.physx.tensors.plugin] Pattern '/World/envs/env_*/Robot' did not match any articulations
2024-01-14 11:43:17 [17,413ms] [Error] [carb.events.python] AttributeError: 'NoneType' object has no attribute 'shared_metatype'

Ltesfaye, 2024-01-19:
This issue also applies in  Create_quadruped_base_env.py  in tutorial 03_envs. When running this with num_envs =1 it generates a different error but works fine if the num_envs is equal to 2. 
`
MySceneCfg(num_envs=1, env_spacing=2.5, replicate_physics=True)
`

It generates the following error: 
```
2024-01-19 20:20:55 [12,020ms] [Warning] [omni.physx.tensors.plugin] Failed to find articulation at '/World/envs/env_0/Robot/base'

2024-01-19 20:20:55 [12,020ms] [Error] [omni.physx.tensors.plugin] Pattern '/World/envs/env_*/Robot/base' did not match any articulations
  ```

Mayankm96, 2024-01-22:
This should be fixed now with this commit: 052c277b7f75288eb7c46edb89524f349b2fdf0a. Please re-open the issue if the problem persists.

closed 2024-01-22 by Mayankm96