isaac-sim/IsaacLab #205: [bug] Camera is not initialized correctly.

catachiii, 2024-01-19:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

I was running the [tutorial of adding sensors to robots](https://isaac-orbit.github.io/orbit/source/tutorials/04_sensors/add_sensors_on_robot.html), and during resetting the camera, robot pos in `_data` will be updated according to the world poses. Until this moment, `self._data` is None, so it gives me error of `TypeError: 'NoneType' object does not support item assignment` during executing `self._data.pos_w[env_ids] = poses`

I was able to trace back to the initialization of the camera class and found `_create_buffers` is never called but `_initialize_impl` is called during `sim.reset()`. 

I assume there is a bug when initializing the camera, but I'm not sure why it happens...

catachiii, 2024-01-19:
Hmmm, I found that it works with the command line command. Not sure why this happens when debugging only.

closed 2024-01-19 by catachiii

cnalty, 2024-04-15:
I'm having a similar issue with contact sensors in my environment. What do you mean it works with a command line command and only breaks in debugging? When I try to run
`orbit.sh -p my_env.py`
I get the following error

> Traceback (most recent call last):
>   File "/home/chris/PycharmProjects/digit_orbit/./envs/digit_torso_env.py", line 275, in <module>
>     main()
>   File "/home/chris/PycharmProjects/digit_orbit/./envs/digit_torso_env.py", line 248, in main
>     env.reset()
>   File "/home/chris/PycharmProjects/digit_orbit/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/base_env.py", line 238, in reset
>     self._reset_idx(indices)
>   File "/home/chris/PycharmProjects/digit_orbit/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/rl_task_env.py", line 312, in _reset_idx
>     self.scene.reset(env_ids)
>   File "/home/chris/PycharmProjects/digit_orbit/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/scene/interactive_scene.py", line 227, in reset
>     sensor.reset(env_ids)
>   File "/home/chris/PycharmProjects/digit_orbit/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/contact_sensor/contact_sensor.py", line 127, in reset
>     self._data.net_forces_w[env_ids] = 0.0
> TypeError: 'NoneType' object does not support item assignment