isaac-sim/IsaacLab #208: [Question] Failure to parse the configuration of grouped articulations in one usd file with ArticulationCfg class

YitianShi, 2024-01-22:
Hi everyone, recently I've built a simple manipulator with robotiq 2f 85 mounted on the end effector of ur10e and tried to load it into the ArticulationCfg class and instantiate a scene with it as a robot. But it seems like only one articulation root is able to be replaced into the scene (As I have two seperate articulation roots as gripper and robot arm).

The following is my usd hierarchy (ManipulatorMounted.usd):  
[image]

My configuration code is:
```
MOUNTED_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path=f"{USD_PATH}/ManipulatorMounted.usd",
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            max_depenetration_velocity=5.0,
        ),
        activate_contact_sensors=False,
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        joint_pos={
            "shoulder_pan_joint": 0,
            "shoulder_lift_joint": -1.2,
            "elbow_joint": 1.1,
            "wrist_1_joint": 0,
            "wrist_2_joint": 0,
            "wrist_3_joint": 0.0,
        },
         pos = (0.5, 1, 0.9)
    ),
    actuators={
        "arm": ImplicitActuatorCfg(
            joint_names_expr=[
            "shoulder_pan_joint",
            "shoulder_lift_joint",
            "elbow_joint",
            "wrist_1_joint",
            "wrist_2_joint",
            "wrist_3_joint"],
            velocity_limit=100.0,
            effort_limit=8700.0,
            stiffness=800.0,
            damping=100.0,
        ),
    },
)
```
Notice that the data are adapted from the UR10 tutorials. 

And configured through:

```
class CellSceneCfg(InteractiveSceneCfg):
    """Configuration for a cell scene."""

    robot: ArticulationCfg = MOUNTED_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot")
```
Following is my RL environment configuration:
```
@configclass
class CellEnvCfg(RLTaskEnvCfg):
    """Configuration for the locomotion velocity-tracking environment."""

    # Scene settings
    scene: CellSceneCfg = CellSceneCfg(num_envs=4096, env_spacing=3.0, replicate_physics=True)
    # Basic settings
    observations: ObservationsCfg = ObservationsCfg()
    actions: ActionsCfg = ActionsCfg()
    randomization: RandomizationCfg = RandomizationCfg()
    # MDP settings
    curriculum: CurriculumCfg = CurriculumCfg()
    rewards: RewardsCfg = RewardsCfg()
    terminations: TerminationsCfg = TerminationsCfg()
    # No command generator
    commands: CommandsCfg = CommandsCfg()

    # Post initialization
    def __post_init__(self) -> None:
        """Post initialization."""
        # general settings
        self.decimation = 2
        self.episode_length_s = 5
        # viewer settings
        self.viewer.eye = (8.0, 0.0, 5.0)
        # simulation settings
        self.sim.dt = 1 / 120
```

It shows the following error if both the robotiq gripper and robot arm are articulation roots:
```
2024-01-21 23:56:04 [15,490ms] [Error] [carb.events.python] RuntimeError: Failed to find a single articulation root when resolving '/World/envs/env_.*/Robot'. Found roots '[Usd.Prim(</World/envs/env_0/Robot/UR10e>), Usd.Prim(</World/envs/env_0/Robot/robotiq_arg2f_85_model>)]' under '/World/envs/env_0/Robot'.
```
It shows the following error if I remove the seperate articulation roots while set the `Root` path as the articulation root:

```
2024-01-21 23:11:42 [17,913ms] [Error] [omni.physx.tensors.plugin] Pattern '/World/envs/env_*/Robot/(base_link|base|base_link_inertia|shoulder_link|upper_arm_link|forearm_link|wrist_1_link|wrist_2_link|wrist_3_link|flange|tool0|robotiq_arg2f_base_link|left_outer_knuckle|left_inner_knuckle|right_inner_knuckle|right_outer_knuckle|left_outer_finger|right_outer_finger|left_inner_finger|right_inner_finger|left_inner_finger_pad|right_inner_finger_pad)' did not match any rigid bodies

2024-01-21 23:11:42 [17,913ms] [Error] [carb.events.python] AttributeError: 'NoneType' object has no attribute 'prim_paths'
```
Does anyone have idea how to mount such a gripper on the robot properly so that it can be parsed by the InteractiveSceneCfg class properly?

SantiDiazC, 2024-04-03:
Hi, I'm trying a similar thing as you describe in your case, I set the robot config as:

``` python
UR5_ROBOTIQ_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        # usd_path= get_assets_root_path() + "/Isaac/Robots/UniversalRobots/ur5/ur5.usd",
        usd_path = "/home/hri-david/Omniverse/Orbit/Santi_tests/assets/ur5_robotiq_2f_140.usd",
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            max_depenetration_velocity=5.0,
        ),
        activate_contact_sensors=False,
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        joint_pos={
            "shoulder_pan_joint": 0.0,
            "shoulder_lift_joint": -1.712,
            "elbow_joint": 1.712,
            "wrist_1_joint": 0.0,
            "wrist_2_joint": 0.0,
            "wrist_3_joint": 0.0,
            "body_f1_.*": 0.0,
            "f1_f2_.*": 0.0,
        },
    ),
    actuators={
        "shoulder": ImplicitActuatorCfg(
            joint_names_expr=["shoulder_.*"],
            velocity_limit=100.0,
            effort_limit=87.0,
            stiffness=800.0,
            damping=40.0,
        ),
        "elbow": ImplicitActuatorCfg(
            joint_names_expr=["elbow_joint"],
            velocity_limit=100.0,
            effort_limit=87.0,
            stiffness=800.0,
            damping=40.0,
        ),
        "wrist": ImplicitActuatorCfg(
            joint_names_expr=["wrist_.*"],
            velocity_limit=100.0,
            effort_limit=87.0,
            stiffness=800.0,
            damping=40.0,
        ),
        "gripper_out": ImplicitActuatorCfg(
            joint_names_expr=["body_f1_.*"],
            velocity_limit=100.0,
            effort_limit=87.0,
            stiffness=800.0,
            damping=40.0,
        ),
        "gripper_in": ImplicitActuatorCfg(
            joint_names_expr=["f1_f2_.*"],
            velocity_limit=100.0,
            effort_limit=87.0,
            stiffness=800.0,
            damping=40.0,
        ),
    },
)
```

and my usd file hierarchy is:

[image]

However, I have a similar error as you: 

``` error
2024-04-03 12:13:35 [12,858ms] [Error] [carb.events.python] RuntimeError: Failed to find a single articulation root when resolving '/World/Origin4/Robot/ur5'. Found roots '[]' under '/World/Origin4/Robot/ur5'.

At:
  /home/hri-david/Omniverse/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py(487): _initialize_impl
  /home/hri-david/Omniverse/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/asset_base.py(255): _initialize_callback
  /home/hri-david/Omniverse/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/asset_base.py(92): <lambda>
  /home/hri-david/Omniverse/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/simulation_context.py(440): render
  /home/hri-david/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py(889): play
  /home/hri-david/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py(592): initialize_physics
  /home/hri-david/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py(624): reset
  /home/hri-david/Omniverse/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/simulation_context.py(369): reset
  /home/hri-david/Omniverse/Orbit/Santi_tests/test_robot_manipulator.py(173): main
  /home/hri-david/Omniverse/Orbit/Santi_tests/test_robot_manipulator.py(182): <module>
```

I wonder if you find a solution to make it work, 

Thank you in advance!

thanyu-hub, 2024-07-22:
I'm wondering how we can combine UR/Panda and Robotiq Gripper. Any updates about this?

Dhoeller19, 2024-10-02:
Hi all,
Unfortunately this is not supported. The way to bypass this for now is to create a separate USD that already combines both articulations (for example using the robot assembler https://docs.omniverse.nvidia.com/isaacsim/latest/advanced_tutorials/tutorial_advanced_assembling_robots.html)