isaac-sim/IsaacLab #209: ModuleNotFoundError: No module named 'omni.isaac.orbit'

DaojiePENG, 2024-01-22:
### Question
I followed the [installation guide](https://isaac-orbit.github.io/orbit/source/setup/installation.html) , but finally, I got this error:

'''
(orbit_env37) pdj@pdj:~/orbit/orbit$ python source/standalone/tutorials/00_sim/create_empty.py
Traceback (most recent call last):
  File "source/standalone/tutorials/00_sim/create_empty.py", line 20, in <module>
    from omni.isaac.orbit.app import AppLauncher
ModuleNotFoundError: No module named 'omni.isaac.orbit'
'''
I can launch `Isaac Sim` with no problems. It seems like I haven't done anything about orbit. I am not sure what I have missed.

DaojiePENG, 2024-01-23:
I realize this is a problem related to python version for I only used the python3.7.

closed 2024-01-23 by DaojiePENG

mohamedamr13, 2024-02-05:
@DaojiePENG Hi, I have the same problem on python 3.10.13. Are you sure this is an issue with the python version ?

Mayankm96, 2024-02-10:
Hi @mohamedamr13 ,

Can you re-open the issue and provide more information?

mohamedamr13, 2024-02-13:
> Hi @mohamedamr13 ,
> 
> Can you re-open the issue and provide more information?

Hi @Mayankm96, this was a simple issue where the extensions where not correctly installed. There was a conflict for the library rl_games with my python and after attempting the reinstall, the import worked right away. Do you still want me to reopen the issue ?

mattmazzola, 2024-05-10:
Related issues:
#406 #407 #103