isaac-sim/IsaacLab #210: [Bug Report] Cant run SKRL example after installation

AirChord, 2024-01-23:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Hi, I'm trying to run the example "Isaac-Reach-Franka-v0" using SKRL library and it returns a error "... KeyError: 'skrl_cfg_entry_point' ". 

### Steps to reproduce
I'm running a conda environment and symply run the command " ./orbit.sh -p source/standalone/workflows/skrl/train.py --task Isaac-Reach-Franka-v0 --headless" and it return the error 
"
2024-01-23 13:32:45 [6,347ms] [Error] [__main__] 'skrl_cfg_entry_point'
2024-01-23 13:32:45 [6,347ms] [Error] [__main__] Traceback (most recent call last):
  File "/home/idt/orbit/source/standalone/workflows/skrl/train.py", line 188, in <module>
    main()
  File "/home/idt/orbit/source/standalone/workflows/skrl/train.py", line 77, in main
    experiment_cfg = load_cfg_from_registry(args_cli.task, "skrl_cfg_entry_point")
  File "/home/idt/orbit/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/utils/parse_cfg.py", line 56, in load_cfg_from_registry
    cfg_entry_point = gym.spec(task_name).kwargs.pop(entry_point_key)
KeyError: 'skrl_cfg_entry_point'

"

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

### System Info

Describe the characteristic of your environment:

- Commit: [9c3f65f3081fc2b51c08e18c7daf97af5c503ac1]
- Isaac Sim Version: [e2023.1.1-rc.8+2023.1.688.573e0291.tc(orbit)]
- OS: [Ubuntu 20.04]
- GPU: [RTX 3070]
- CUDA: [12.3]
- GPU Driver: [525.147.05]

### Additional context

Add any other context about the problem here.

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

AirChord, 2024-02-02:
The skrl agents are missing in all manipulation environments, and the entry point is not registered. I followed the installation guide, but can't find the files. Do I need to install anything else?

Mayankm96, 2024-02-10:
Hi,

Due to the recent refactorings, SKRL integration got broken. @Toni-SM is working on fixing this up.

Mayankm96, 2024-02-29:
Hopefully this is now fixed with #249 . Please re-open or create a new issue if the problem persists.

closed 2024-02-29 by Mayankm96