isaac-sim/IsaacLab #219: [Bug Report] Imitation learning tutorial error

MorningFrog, 2024-01-29:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

I am following documents, in [`Getting Started/Running existing scripts/Workflows/Imitation Learning`](https://isaac-orbit.github.io/orbit/source/setup/sample.html#imitation-learning) section had a problem. 

### Steps to reproduce

When I run 
```
./orbit.sh -p source/standalone/workflows/robomimic/collect_demonstrations.py --task Isaac-Lift-Cube-Franka-IK-Rel-v0 --num_envs 1 --num_demos 10 --device keyboard
```
there is an error:
```
[8.724s] Simulation App Startup Complete
[INFO]: Parsing configuration from: <class 'omni.isaac.orbit_tasks.manipulation.lift.config.franka.ik_rel_env_cfg.FrankaCubeLiftEnvCfg'>
2024-01-29 14:33:46 [11,945ms] [Error] [__main__] 'FrankaCubeLiftEnvCfg' object has no attribute 'control'
2024-01-29 14:33:46 [11,945ms] [Error] [__main__] Traceback (most recent call last):
  File "/path/to/Orbit/source/standalone/workflows/robomimic/collect_demonstrations.py", line 173, in <module>
    main()
  File "/path/to/Orbit/source/standalone/workflows/robomimic/collect_demonstrations.py", line 74, in main
    env_cfg.control.control_type = "inverse_kinematics"
AttributeError: 'FrankaCubeLiftEnvCfg' object has no attribute 'control'

[12.134s] Simulation App Shutting Down
```

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [2023.1.1]
- OS: [Ubuntu 20.04]
- GPU: [RTX 3080 Ti]
- CUDA: [12.0]
- GPU Driver: [525.147.05]

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2024-02-10:
This seems like a duplicate of: https://github.com/NVIDIA-Omniverse/orbit/issues/228

We are working on fixing the IL workflow.

Nimingez, 2024-02-29:
When do you think the bug fix will be completed?

jsmith-bdai, 2024-02-29:
@Nimingez We are shooting to fix this in the next week. Sorry for delay!

Nimingez, 2024-03-01:
> @Nimingez We are shooting to fix this in the next week. Sorry for delay!
I want to use spacemouse to collect demonstrations. Just --device spacemouse?

jsmith-bdai, 2024-03-14:
Hi @Nimingez, sorry for the delay once again. We've been actively working on this but there were more API changes on the robomimic side that we are now working through.

Assuming all goes well, we should have in the coming week - I will keep you posted.

Mayankm96, 2024-04-12:
This should be fixed now. It is a bit hard-coded for the lift environment, but it can be adapted easily for your usecases.

We are working on making it more general. But that will take time from our side due to other priorities.

closed 2024-04-12 by Mayankm96