isaac-sim/IsaacLab #228: [Bug Report] AttributeError: 'FrankaReachEnvCfg' object has no attribute 'control'

AliBuildsAI, 2024-02-03:
### Describe the bug
When I run `collect_demonstration.py`, I get the following error:
`AttributeError: 'FrankaReachEnvCfg' object has no attribute 'control'`

(Note:  I checked the code and the environemnets actually don't have the attribute control. They used to have it before the major updates, and it seems like this scripts has not been modified accordingly.)

### Steps to reproduce

`./orbit.sh -p source/standalone/workflows/robomimic/collect_demonstrations.py --task Isaac-Reach-Franka-v0`

Here is the full log
```
cd /home/ali/repos/orbit ; /usr/bin/env /home/ali/miniconda3/envs/orbit24/bin/python /home/ali/.vscode/extensions/ms-python.python-2023.22.1/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher 36157 -- /home/ali/repos/orbit/source/standalone/workflows/robomimic/collect_demonstrations.py 
[INFO][AppLauncher]: Loading experience file:  .
Starting kit application with the following args:  ['/home/ali/repos/orbit/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/ali/repos/orbit/_isaac_sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/home/ali/repos/orbit/_isaac_sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/ali/repos/orbit/_isaac_sim/exts', '--ext-folder', '/home/ali/repos/orbit/_isaac_sim/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  []
[Info] [carb] Logging to file: /home/ali/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/logs/Kit/Isaac-Sim/2023.1/kit_20240202_170500.log
2024-02-03 01:05:00 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.098s] [ext: omni.kit.async_engine-0.0.0] startup
[1.052s] [ext: omni.activity.core-1.0.1] startup
[1.056s] [ext: omni.assets.plugins-0.0.0] startup
[1.057s] [ext: omni.stats-0.0.0] startup
[1.058s] [ext: omni.client-1.0.1] startup
[1.064s] [ext: omni.activity.profiler-1.0.2] startup
[1.066s] [ext: omni.gpu_foundation-0.0.0] startup
[1.075s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[1.077s] [ext: carb.windowing.plugins-1.0.0] startup
[1.091s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 525.147.05    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24810   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | ee556d89.. |
|     |                                  |        |     |            | 65        |            |
|=============================================================================================|
| OS: 20.04.6 LTS (Focal Fossa) ubuntu, Version: 20.04.6, Kernel: 5.15.0-92-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: Intel(R) Core(TM) i9-10900X CPU @ 3.70GHz | Cores: 10 | Logical: 20
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 63979 | Free Memory: 54675
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
[1.792s] [ext: omni.kit.pipapi-0.0.0] startup
[1.794s] [ext: omni.kit.pip_archive-0.0.0] startup
[1.794s] [ext: omni.pip.compute-1.2.0] startup
[1.796s] [ext: omni.pip.cloud-1.0.1] startup
[1.798s] [ext: omni.isaac.core_archive-2.2.1] startup
[1.799s] [ext: omni.pip.torch-2_0_1-2.0.2] startup
[1.867s] [ext: omni.kit.telemetry-0.5.0] startup
[1.888s] [ext: omni.kit.loop-isaac-1.1.0] startup
[1.889s] [ext: omni.kit.test-0.0.0] startup
[1.891s] [ext: omni.isaac.ml_archive-1.1.3] startup
[1.891s] [ext: omni.mtlx-0.1.0] startup
[1.892s] [ext: omni.usd.config-1.0.3] startup
[1.898s] [ext: omni.gpucompute.plugins-0.0.0] startup
[1.898s] [ext: omni.usd.libs-1.0.0] startup
[2.031s] [ext: omni.appwindow-1.1.5] startup
[2.691s] [ext: omni.kit.renderer.core-0.0.0] startup
[5.001s] [ext: omni.kit.renderer.capture-0.0.0] startup
[5.004s] [ext: omni.kit.renderer.imgui-0.0.0] startup
[5.170s] [ext: carb.audio-0.1.0] startup
[5.192s] [ext: omni.ui-2.18.5] startup
[5.219s] [ext: omni.uiaudio-1.0.0] startup
[5.221s] [ext: omni.kit.mainwindow-1.0.1] startup
[5.224s] [ext: omni.kit.uiapp-0.0.0] startup
[... 286 lines cut ...]
2024-02-03 01:07:47 [167,307ms] [Warning] [omni.core.ITypeFactory] Module /home/ali/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
```

For more information on this, check: https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

-->

### System Info

Describe the characteristic of your environment:

- Commit: 670a8c97db
- Isaac Sim Version: 2023.1.1-rc.8+2023.1.688.573e0291.tc(base)
- OS: Ubuntu 20.04
- GPU: RTX 4090
- CUDA: 12.3
- GPU Driver: 525.147.05
### Additional context

Add any other context about the problem here.

### Checklist

- I have checked that there is no similar issue in the repo (**required**)
- I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2024-02-10:
Thank you for bringing this up. Seems like the workflow script was not updated with the latest refactoring. We are working now on fixing this.

Mayankm96, 2024-04-12:
This should be fixed now.

closed 2024-04-12 by Mayankm96