isaac-sim/IsaacLab #233: [Bug Report] Ray Caster Camera not working properly

samibouziri, 2024-02-06:
### Describe the bug

The Ray Caster Camera does not work properly. It creates hitting points in the wrong places. 

### Steps to reproduce
A way to reproduce the bug is to run the Ray Caster Camera tutorial with one camera:
./orbit.sh -p source/standalone/tutorials/04_sensors/run_ray_caster_camera.py
The outputted points are not always on the rough terrain.
More over :
- When replacing the terrain in the mesh prim path with a shape config we get an error.
- When replacing the terrain in the mesh prim path with with the table that we have in other tutorials we get an error.
- When replacing the terrain in the mesh prim path with "{ISAAC_NUCLEUS_DIR}/Props/YCB/Axis_Aligned_Physics/003_cracker_box.usd", there are no errors but the point cloud is under the terrain and none of the points are on the box.

Note: I have also tried to draw the point cloud myself following the procedure given in the documentation and had the same result. 

### System Info

Describe the characteristic of your environment:

- Commit: 670a8c97db645d50a3ea89b06f7366fc2212fc06
- Isaac Sim Version: 2023.1.0-rc.42+2023.1.667.d641947f.tc
- OS: Ubuntu 22.04
- GPU: RTX 3090 
- CUDA: 12.2
- GPU Driver: 535.113.01

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo (To my knowledge ray caster is only implemented in orbit. I haven't verified if the warp works correctly or not)

samibouziri, 2024-02-07:
FYI this is what you get for the default plane cfg:
[image]
and this is what you get when placing the pov camera at the same at the same position and orientation as the raycaster camera for the rough terrain like in the tutorial (expected behavious a red rectangle):
[image]
as you can see we get the expected behavior (more or less. it is shifted a bit as you can see in the second photo) for positive x,y

pascal-roth, 2024-02-07:
> ./orbit.sh -p source/standalone/tutorials/04_sensors/run_ray_caster_camera.py
> The outputted points are not always on the rough terrain.

The `rough_plane.usd` of Nuclues has a translation of (40, 40, 0), if you remove it, the output fits. In our minimal example, we tested with our local version that does not have such an offset. It is not a fault of the raycaster. Will adapt the version so that it fits to the Nucleus asset

samibouziri, 2024-02-07:
So here is the full modified code to cast the point cloud on the box 
```# Copyright (c) 2022-2024, The ORBIT Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

"""
This script shows how to use the ray-cast camera sensor from the Orbit framework.

The camera sensor is based on using Warp kernels which do ray-casting against static meshes.

.. code-block:: bash

    # Usage
    ./orbit.sh -p source/standalone/tutorials/04_sensors/run_ray_caster_camera.py

"""

"""Launch Isaac Sim Simulator first."""

import argparse

from omni.isaac.orbit.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="This script demonstrates how to use the ray-cast camera sensor.")
parser.add_argument("--num_envs", type=int, default=16, help="Number of environments to generate.")
parser.add_argument("--save", action="store_true", default=False, help="Save the obtained data to disk.")
# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()
# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

"""Rest everything follows."""

import os
import torch
import traceback

import carb
import omni.isaac.core.utils.prims as prim_utils
import omni.replicator.core as rep

import omni.isaac.orbit.sim as sim_utils
from omni.isaac.orbit.sensors.ray_caster import RayCasterCamera, RayCasterCameraCfg, patterns
from omni.isaac.orbit.utils import convert_dict_to_backend
from omni.isaac.orbit.utils.assets import ISAAC_NUCLEUS_DIR
from omni.isaac.orbit.utils.math import project_points, unproject_depth

def define_sensor() -> RayCasterCamera:
    """Defines the ray-cast camera sensor to add to the scene."""
    # Camera base frames
    # In contras to the USD camera, we associate the sensor to the prims at these locations.
    # This means that parent prim of the sensor is the prim at this location.
    prim_utils.create_prim("/World/Origin_00/CameraSensor", "Xform")
    # prim_utils.create_prim("/World/Origin_01/CameraSensor", "Xform")

    # Setup camera sensor
    camera_cfg = RayCasterCameraCfg(
        prim_path="/World/Origin_.*/CameraSensor",
        mesh_prim_paths=["/World/Objects/Box"],
        update_period=0.1,
        offset=RayCasterCameraCfg.OffsetCfg(pos=(0.0, 0.0, 0.0), rot=(1.0, 0.0, 0.0, 0.0)),
        data_types=["distance_to_image_plane", "normals", "distance_to_camera"],
        debug_vis=True,
        pattern_cfg=patterns.PinholeCameraPatternCfg(
            focal_length=24.0,
[... 107 lines cut ...]
    # Play simulator
    sim.reset()
    # Now we are ready!
    print("[INFO]: Setup complete...")
    # Run simulator
    run_simulator(sim=sim, scene_entities=scene_entities)

if __name__ == "__main__":
    try:
        # run the main execution
        main()
    except Exception as err:
        carb.log_error(err)
        carb.log_error(traceback.format_exc())
        raise
    finally:
        # close sim app
        simulation_app.close()
 ```

and here is the output above the ground:
[image]

and this is the output under the ground:
[image]

samibouziri, 2024-02-07:
@pascal-roth  From the comment above I understand now what's happening here. Basically same issue as the ground but instead of having some translation issues we have a scaling issue.

pascal-roth, 2024-02-07:
No worries. These things are actually already in a PR and should be merged soon.

closed 2024-02-08 by samibouziri

gongceerhao, 2024-09-05:
[image]

gongceerhao, 2024-09-05:
What happen？

samibouziri, 2024-09-05:
Well, if nothing changed from my last post,  you can only raycast on one static asset. And the raycaster only considers the mesh of that asset in it s default state. 
For boxes the default state should be a box centered around (0,0,0). So no matter where you translate the box, if it is not done in the creation of that mesh (which cannot be done using isaac lab functionalities; in isaac lab you first create the asset around 0,0,0 and then it is translated during the initialization of the sim), the raycaster will not perceived that translation. 
To percieve the translation you should first create the Mesh using another software and translate it there and then in isaac lab create an asset from that Mesh.