isaac-sim/IsaacLab #235: [Question] Where should I put my own library in orbit?

fan-ziqi, 2024-02-08:
### Question

Where should I put my own library like [RL-Games](https://github.com/Denys88/rl_games) or [RSL-RL](https://github.com/leggedrobotics/rsl_rl) in orbit?

Mayankm96, 2024-02-10:
Hi,

These libraries are installed as Python modules when you run `./orbit.sh -e`. You can integrate your library in a similar way by the following steps:

1. Add your library as a dependency for installation: https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/extensions/omni.isaac.orbit_tasks/setup.py#L34-L41
2. Install your library by running `./orbit.sh -e "YOU_LIB_NAME"`
3. Create a wrapper for the library (in-case it doesn't follow gym.Env convention): https://github.com/NVIDIA-Omniverse/orbit/tree/main/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/utils/wrappers
4. (optional) Add some tests to make sure your wrappers work: https://github.com/NVIDIA-Omniverse/orbit/tree/main/source/extensions/omni.isaac.orbit_tasks/test/wrappers
5. Create workflow scripts for your library to train and play the trained agents using your library: https://github.com/NVIDIA-Omniverse/orbit/tree/main/source/standalone/workflows

If you want to use your library from source, then instead of doing (1) and (2), you can just install your library similar to other Python libraries. As an example:

```bash
# Option 1: You are using a conda environment with Orbit
cd PATH/TO/YOUR_LIB_DIR
python -m pip install -e .

# Option 2: You are using the Python shipped with Isaac Sim
cd orbit
./orbit.sh -p -m pip install -e PATH/TO/YOUR_LIB_DIR
```

fan-ziqi, 2024-02-11:
Thank you very much!

closed 2024-02-11 by fan-ziqi