isaac-sim/IsaacLab #236: [Bug Report] RayCaster Failing - no spawn function available

WillMandil001, 2024-02-08:
Hello All,

I am trying to use the ray_caster sensor (for a custom version of the lift environment), However, I get issues as the initialize function for the raycaster or the raycastercamera are unable to find the prim (under /World/env/env_*/height_scanner) - I assume this is because, unlike the camera, there is no spawn option for the raycaster:

Here is how I am creating the raycaster in lift_env_cfg.py:
`
@configclass
class ObjectTableSceneCfg(InteractiveSceneCfg):
    # robots: will be populated by agent env cfg
    robot: ArticulationCfg = MISSING
    # end-effector sensor: will be populated by agent env cfg
    ee_frame: FrameTransformerCfg = MISSING
    # target object: will be populated by agent env cfg
    object: RigidObjectCfg = MISSING

    if _use_obstacles:
        # obstacles: will be populated by agent env cfg
        obstacles: RigidObjectCfg = MISSING

    # Table
    table = AssetBaseCfg( prim_path="{ENV_REGEX_NS}/Table", init_state=AssetBaseCfg.InitialStateCfg(pos=[0.5, 0, 0], rot=[0.707, 0, 0, 0.707]), spawn=UsdFileCfg(usd_path=f"{ISAAC_NUCLEUS_DIR}/Props/Mounts/SeattleLabTable/table_instanceable.usd"), )

    # plane
    plane = AssetBaseCfg( prim_path="/World/GroundPlane", init_state=AssetBaseCfg.InitialStateCfg(pos=[0, 0, -1.05]), spawn=GroundPlaneCfg(),)

    # lights
    light = AssetBaseCfg( prim_path="/World/light", spawn=sim_utils.DomeLightCfg(color=(0.75, 0.75, 0.75), intensity=3000.0), )

        height_scanner = RayCasterCfg(
            prim_path="{ENV_REGEX_NS}/height_scanner",
            mesh_prim_paths=["/World/defaultGroundPlane"],
            offset=RayCasterCfg.OffsetCfg(pos=(1.6, 1.2, 1.3), rot=(-0.17591989, 0.33985114, 0.82047325, -0.42470819)),
            pattern_cfg=patterns.GridPatternCfg(resolution=0.1, size=(2.0, 2.0)),
            attach_yaw_only=True,
            debug_vis=True,
        )
`

and the observations are created as such: 

`
@configclass
class ObservationsCfg:

    @configclass
    class PolicyCfg(ObsGroup):
        """Observations for policy group."""

        joint_pos = ObsTerm(func=mdp.joint_pos_rel)
        joint_vel = ObsTerm(func=mdp.joint_vel_rel)

        object_position = ObsTerm(func=mdp.object_position_in_robot_root_frame)
    
        target_object_position = ObsTerm(func=mdp.generated_commands, params={"command_name": "object_pose"})

        actions = ObsTerm(func=mdp.last_action)

        # iamges = ObsTerm(func=mdp.camera_image, params={"camera_name": "front_cam", "data_type": "rgb"})

        def __post_init__(self):
            self.enable_corruption = True
            self.concatenate_terms = True

    if _use_camera:
        @configclass
        class DepthCameraCfg(ObsGroup):
            """Observations for policy group."""

            depthimages = ObsTerm(func=mdp.depth_camera_image, params={"camera_name": "front_cam", "data_type": "rgb"})

            def __post_init__(self):
                self.enable_corruption = True
                self.concatenate_terms = True

        depth: DepthCameraCfg = DepthCameraCfg()
    
    if _use_raycast:
        @configclass
        class RayCasterCameraCfg(ObsGroup):
            """Observations for policy group."""

            raycasterimages = ObsTerm(func=mdp.ray_caster_image, params={"camera_name": "height_scanner", "data_type": "ray_caster_image"})

            def __post_init__(self):
                self.enable_corruption = True
                self.concatenate_terms = True

        ray_caster: RayCasterCameraCfg = RayCasterCameraCfg()

    # observation groups
    policy: PolicyCfg = PolicyCfg()
`

The error occurs under the ray_caster.py where it looks for the prim: (which is not spawned as there is no spawn function in this script - unlike the camera.py script, which spawns the camera on __init__)
`
    def _initialize_impl(self):
        super()._initialize_impl()
        # create simulation view
        self._physics_sim_view = physx.create_simulation_view(self._backend)
        self._physics_sim_view.set_subspace_roots("/")
        # check if the prim at path is an articulated or rigid prim we do this since for physics-based view classes we can access their data directly otherwise we need to use the xform view class which is slower
        found_supported_prim_class = False
        prim = sim_utils.find_first_matching_prim(self.cfg.prim_path)
        if prim is None:
            raise RuntimeError(f"Failed to find a prim at path expression: {self.cfg.prim_path}")
`

Please let me know if I have done something wrong or is this something other people have found?

Describe the characteristic of your environment:
- Isaac Sim Version: 2023.1
- OS: [e.g. Ubuntu 20.04]
- GPU: GeForce RTX 2080 Ti
- CUDA: 12.0
- GPU Driver: 525.60.13

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

samibouziri, 2024-02-09:
I had a similar issue when using the ray caster camera. From my understanding the ray_caster cannot / does not create a prim. So to 'fix' that you should fix the ray caster to an existing prim like the table or the Robot base and then play with the Offset to put the ray caster where you want. An example would be 
```
height_scanner = RayCasterCfg(
            prim_path="{ENV_REGEX_NS}/Table",
            mesh_prim_paths=["/World/defaultGroundPlane"],
            offset=RayCasterCfg.OffsetCfg(pos=(1.6, 1.2, 1.3), rot=(-0.17591989, 0.33985114, 0.82047325, -0.42470819)),
            pattern_cfg=patterns.GridPatternCfg(resolution=0.1, size=(2.0, 2.0)),
            attach_yaw_only=True,
            debug_vis=True,
        )
```

WillMandil001, 2024-02-09:
Wow! That was such an easy fix! thanks so much!!! it works. Have a good weekend!

closed 2024-02-10 by Mayankm96