isaac-sim/IsaacLab #237: [Bug Report] Articulation initialization fails to reconstruct correct paths for its rigid bodies on assets with deep hierarchy

ozhanozen, 2024-02-09:
### Describe the bug

When an Articulation is initialized,  `_initilize_imp()` function tries to construct the body paths by simply combining the root prim path and body names within the articulation tree:

https://github.com/NVIDIA-Omniverse/orbit/blob/477cd6b3f6f37bdf8d7d8a47bf6ea8bb15376c3b/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py#L490-L502

However, this leads to incorrect paths for the bodies that are not right under the root prim. 

For example, if there is an Xform to group a few bodies together:

    robot (defaultPrim)
    --- body1
    --- body2
    --- Xform
    -------- body3 
    -------- body4
the prim path for body3 is assumed to be on /robot/body3 rather than /robot/Xform/body3.

Eventually, the RigidBodyView is not constructed correctly, and this raises a RuntimeError at the following lines due to the mismatch of the `body_names` with the `physx_body_names`:

https://github.com/NVIDIA-Omniverse/orbit/blob/477cd6b3f6f37bdf8d7d8a47bf6ea8bb15376c3b/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py#L510-L512

### Steps to reproduce

Using any Articulation with a similar hierarchy reproduces the error.

### System Info

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [2023.1.0-hotfix.1]
- OS: [Ubuntu 20.04]
- GPU: [e.g. RTX A6000]
- CUDA: [12.0]
- GPU Driver: [525.60.11]

### Additional context

#200 and #208 are probably related to this.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2024-02-10:
Thank you for bringing this up.

We are aware of this issue and have reported it to the PhysX team to fix the regex parser on their side. Currently, the immediate fix is to ensure the asset follows a flat hierarchy of rigid links.

ozhanozen, 2024-02-10:
Thanks for the workarounds!

ozhanozen, 2024-02-10:
> Thank you for bringing this up.
> 
> We are aware of this issue and have reported it to the PhysX team to fix the regex parser on their side. The immediate fix is to ensure the asset follows a flat hierarchy of rigid links.
> 
> If you don't care about all the bodies in the articulation, a more temporary solution is to replace the regex expression on the line you highlighted with the bodies you care about. Although ugly, hopefully, it lets you use the class.

Thanks for the workarounds!

axelgoedrich, 2024-05-20:
Hi @Mayankm96,
I am having the same issue @ozhanozen described.
I tried to replace the regex expression with multiple different versions, but none of them worked (ChatGPT couldn`t help me either).
For example: "/robot/(body1|body2)" works, but when I use "/robot/(body1|body2|Xform/body3)" I get the error Pattern '/robot/(body1|body2|Xform/body3)' did not match any rigid bodies.

The interpretation of the regex expression seems to be very restrictive, or am I missing something?

Any help would be appreciated.

PR #643 (2024-07-05): Removes the use of body view inside the asset classes
# Description

The deprecation notice has been out for a while. The MR removes the body PhysX view from the rigid object and articulation classes. This is no longer needed as their respective root views expose all the data needed for them.

This is a breaking change, and we lose compatibility with Isaac Sim 2023.1.1.

Fixes #237

## Type of change

- Bug fix (non-breaking change which fixes an issue)
- Breaking change (fix or feature that would cause existing functionality to not work as expected)
- This change requires a documentation upda [... cut]
commit da34688b (2024-07-05): Removes the use of body view inside the asset classes (#643)

# Description

The deprecation notice has been out for a while. The MR removes the body
PhysX view from the rigid object and articulation classes. This is no
longer needed as their respective root views expose all the data needed
for them.

This is a breaking change, and we lose compatibility with Isaac Sim
2023.1.1.

Fixes #237

## Type of change

- Bug fix (non-breaking change which fixes an issue)
- Breaking change (fix or feature that would cause existing
functionality to not work as expected)
- This change requires a documentat [... cut]
closed 2024-07-05 by Mayankm96