isaac-sim/IsaacLab #238: [Bug Report] Float object is not subscriptable error in _log_articulation_joint_info() on a 1-DoF Articulation

ozhanozen, 2024-02-10:
### Describe the bug

On a 1-DoF Articulation, the following lines do not create lists but only floats:

https://github.com/NVIDIA-Omniverse/orbit/blob/939a768357db47182c5ea5d6f4efbff5f3e3b433/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py#L705-L713

This creates an error while trying to subscribe to these floats:

https://github.com/NVIDIA-Omniverse/orbit/blob/939a768357db47182c5ea5d6f4efbff5f3e3b433/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py#L736-L742

### Steps to reproduce

Any Articulation with 1 joint creates the error.

### System Info

- Commit: [939a768]
- Isaac Sim Version: [2023.1.0-hotfix.1]
- OS: [Ubuntu 20.04]
- GPU: [RTX A6000]
- CUDA: [12.0]
- GPU Driver: [525.60.11]

### Additional context

Currently using dummy joints on the assets as a temporary fix.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2024-02-12:
Thank you for pointing this out! I am unsure why we did a `squeeze()` operator there. It isn't needed from what I see right now.

If you remove it, does the code then work for you?

Mayankm96, 2024-02-12:
Okay, I managed to repro the issue. If you remove the `squeeze()` operator, it works fine. The fix should hopefully land tomorrow (under review right now).

Thank you again for pointing out this issue!

ozhanozen, 2024-02-13:
> Okay, I managed to repro the issue. If you remove the `squeeze()` operator, it works fine. The fix should hopefully land tomorrow (under review right now).
> 
> Thank you again for pointing out this issue!

It works for me as well like this, thanks a lot!

commit fcc216a5 (2024-03-28): Adds custom experience file loading to the AppLauncher (#473)

# Description

This MR automates the configuring of kit app files based on the settings
passed to the app launcher.

The experience file to load when launching the SimulationApp. If a
relative path is provided, it is resolved relative to the ``EXP_PATH``
environment variable. If provided as an empty string, the experience
file is determined based on the headless flag:

* If headless is True, the experience file is set to
``orbit.python.headless.kit``.
* If headless is False, the experience file is set to
``orbit.python.kit``.

More [... cut]
closed 2024-02-14 by Mayankm96