isaac-sim/IsaacLab #239: sys.stdout becomes omni.kit.app._impl.StdOutInterceptor when training with --headless

zoctipus, 2024-02-11:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

```
2024-02-11 09:36:26 [912,695ms] [Error] [__main__] Expected file or str, got <omni.kit.app._impl.StdOutInterceptor object at 0x7fde081ea080>
2024-02-11 09:36:26 [912,695ms] [Error] [__main__] Traceback (most recent call last):
  File "/home/octipus/Projects/orbit/tychoorbit/workflows/sb3/train_SAC.py", line 146, in <module>
    main()
  File "/home/octipus/Projects/orbit/tychoorbit/workflows/sb3/train_SAC.py", line 129, in main
    new_logger = configure(log_dir, ["stdout", "tensorboard"])
  File "/home/octipus/anaconda3/envs/tychoorbit/lib/python3.10/site-packages/stable_baselines3/common/logger.py", line 657, in configure
    output_formats = [make_output_format(f, folder, log_suffix) for f in format_strings]
  File "/home/octipus/anaconda3/envs/tychoorbit/lib/python3.10/site-packages/stable_baselines3/common/logger.py", line 657, in <listcomp>
    output_formats = [make_output_format(f, folder, log_suffix) for f in format_strings]
  File "/home/octipus/anaconda3/envs/tychoorbit/lib/python3.10/site-packages/stable_baselines3/common/logger.py", line 455, in make_output_format
    return HumanOutputFormat(sys.__stdout__)
  File "/home/octipus/anaconda3/envs/tychoorbit/lib/python3.10/site-packages/stable_baselines3/common/logger.py", line 171, in __init__
    raise ValueError(f"Expected file or str, got {filename_or_file}")
ValueError: Expected file or str, got <omni.kit.app._impl.StdOutInterceptor object at 0x7fde081ea080>
```

I am not sure under what condition, but adding the key word `--headless` in training a sb3 agent in my case will make `sys.std` be a `<omni.kit.app._impl.StdOutInterceptor object at 0x7fde081ea080> `  instead of  `<_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>`, and causes loggers to Error`raise ValueError(f"Expected file or str, got {filename_or_file}")`. Interestly removing --headless will also remove will be alright. How to make sure that sys.std always return to the standard `<_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>`>

For more information on this, check: https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

-->

### System Info

Describe the characteristic of your environment:

- Isaac Sim Version: 2023.1.0
- OS:  Ubuntu 22.04
- GPU: RTX 4090

### Checklist

- I have checked that there is no similar issue in the repo (**required**)
- I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1. sys.std is always `<_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>` when creating the logger in sb3 so the logger is successfully created when training is initialized

lorandcheng, 2024-02-16:
Also getting this error, the [cartpole RL headless example](https://isaac-orbit.github.io/orbit/source/tutorials/03_envs/run_rl_training.html) throws the same thing

zoctipus, 2024-03-18:
```
def make_output_format(_format: str, log_dir: str, log_suffix: str = "") -> KVWriter:
    """
    return a logger for the requested format

    :param _format: the requested format to log to ('stdout', 'log', 'json' or 'csv' or 'tensorboard')
    :param log_dir: the logging directory
    :param log_suffix: the suffix for the log file
    :return: the logger
    """
    os.makedirs(log_dir, exist_ok=True)
    if _format == "stdout":
        return HumanOutputFormat(sys.stdout)
```

in above code located in` site-packages/stable_baseline3/common/logger.py    line 455`, I can mannually change `sys.stdout` to `sys.__stdout__` to get the right iowriter back hopefully there will be a better solution

Mayankm96, 2024-03-28:
Thank you for reporting this. It turned out that Isaac Sim 2023.1.1 had a different app settings (from Isaac Sim 2023.1.0), which led to this issue.

We made a fix for this through fcc216a5399661b6f3125e1e0a99235f073a3124 . I checked it locally and it works fine for me.

The main changes needed for it are as follows:

> https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/apps/orbit.python.headless.kit#L80-L83

closed 2024-03-28 by Mayankm96