isaac-sim/IsaacLab #242: check_instanceable.py in /source/standalone/tools has issues

phenomen21, 2024-02-13:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

In line 106 of the check_instanceable.py script there is an error - line goes as
``
prim_utils.create_prim("/World/envs/env_0/Asset", "Xform", usd_path=os.path.abspath(args_cli.usd_path))
``
but it should be
``
prim_utils.create_prim("/World/envs/env_0/Asset", "Xform", usd_path=os.path.abspath(args_cli.input))
``

### Steps to reproduce

Run check_instanceable.py script with any USD file.

### System Info

Describe the characteristic of your environment:

- Commit: 963f304
- Isaac Sim Version: 2023.1.1-rc .8+2023.1.688.573e0291.tcroot
- OS: Ubuntu 22.04
- GPU: Nvidia A10
- CUDA: 12.0
- GPU Driver: 525.85.05

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] check_instanceable.py should work without crashing

Mayankm96, 2024-02-14:
Thank you for pointing this out. Would you like to send the MR with the fix? :)

phenomen21, 2024-02-14:
Ok will do

PR #248 (2024-02-29): Fixes typo in the check_instanceable.py script
# Description

In line 106 of the `check_instanceable.py` script, there is a typo error.

The line goes as:

```python
prim_utils.create_prim("/World/envs/env_0/Asset", "Xform", usd_path=os.path.abspath(args_cli.usd_path))
```

but should be:

```python
prim_utils.create_prim("/World/envs/env_0/Asset", "Xform", usd_path=os.path.abspath(args_cli.input))
```

Fixes #242 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./orbit.sh --format [... cut]
commit a0274e49 (2024-02-29): Fixes typo in the check_instanceable.py script (#248)

# Description

In line 106 of the `check_instanceable.py` script, there is a typo
error.

The line goes as:

```python
prim_utils.create_prim("/World/envs/env_0/Asset", "Xform", usd_path=os.path.abspath(args_cli.usd_path))
```

but should be:

```python
prim_utils.create_prim("/World/envs/env_0/Asset", "Xform", usd_path=os.path.abspath(args_cli.input))
```

Fixes #242 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./orbit.sh  [... cut]
closed 2024-02-29 by Mayankm96