isaac-sim/IsaacLab #247: [Question] FileNotFoundError when running the extensions orbit_tasks for training locomotion

dnjsxor999, 2024-02-18:
I am trying to run the locomotion extensions as an example with training.py in rsl_rl and task: “Isaac-Velocity-Flat-Anymal-B-v0”
so I command like this:
`./orbit.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Velocity-Flat-Anymal-B-v0`

but error shows up after Isaacsim simulation immediately ends.
```
[6.050s] Simulation App Startup Complete
Traceback (most recent call last):
File “/home/wt/.local/share/ov/pkg/isaac_sim-2023.1.1/Orbit/source/standalone/workflows/rsl_rl/train.py”, line 64, in
import omni.isaac.orbit_tasks # noqa: F401
File “/home/wt/anaconda3/envs/orbit_env/lib/python3.10/site-packages/omni/isaac/orbit_tasks/init.py”, line 15, in
ORBIT_TASKS_METADATA = toml.load(os.path.join(ORBIT_TASKS_EXT_DIR, “config”, “extension.toml”))
File “/home/wt/.local/share/ov/pkg/isaac_sim-2023.1.1/Orbit/_isaac_sim/kit/exts/omni.kit.pip_archive/pip_prebundle/toml/decoder.py”, line 133, in load
with io.open(_getpath(f), encoding=‘utf-8’) as ffile:
FileNotFoundError: [Errno 2] No such file or directory: ‘/home/wt/anaconda3/envs/orbit_env/lib/python3.10/site-packages/config/extension.toml’
2024-02-18 00:33:15 [7,344ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/wt/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
2024-02-18 00:33:15 [7,346ms] [Warning] [omni.stageupdate.plugin] Deprecated: direct use of IStageUpdate callbacks is deprecated. Use IStageUpdate::getStageUpdate instead.
2024-02-18 00:33:15 [7,347ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2024-02-18 00:33:15 [7,347ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2024-02-18 00:33:15 [7,347ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2024-02-18 00:33:15 [7,347ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2024-02-18 00:33:15 [7,347ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2024-02-18 00:33:15 [7,347ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2024-02-18 00:33:15 [7,347ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2024-02-18 00:33:15 [7,347ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2024-02-18 00:33:15 [7,347ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2024-02-18 00:33:15 [7,347ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2024-02-18 00:33:15 [7,347ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Annotators’ for removal
2024-02-18 00:33:15 [7,347ms] [Warning] [omni.graph.core.plugin] Could not find category ‘Replicator:Core’ for removal
2024-02-18 00:33:15 [7,349ms] [Warning] [carb.audio.context] 1 contexts were leaked
2024-02-18 00:33:15 [7,368ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2024-02-18 00:33:15 [7,375ms] [Warning] [omni.core.ITypeFactory] Module /home/wt/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
```
I can see that there is the config/extension.toml file in the extensions/omni.isaac.orbit_tasks, but I don’t know why this “File Not Found” happens.

Could you figure it out? or did I miss something?

Mayankm96, 2024-02-18:
Are you able to run the demo or tutorial scripts?

Mayankm96, 2024-07-29:
Closing this issue due to lack of activity. Please re-open if the problem persists.

closed 2024-07-29 by Mayankm96