isaac-sim/IsaacLab #252: [Bug and improvement] Small typo in utils.math

samibouziri, 2024-02-21:
in this line:
https://github.com/NVIDIA-Omniverse/orbit/blob/7a967894946cf6cf52040c8def90892d04856eea/source/extensions/omni.isaac.orbit/omni/isaac/orbit/utils/math.py#L637C1-L637C50
you can either change the dim=-1 or change the  documentation of the function to only accept (N,4) and return (N,3).

If you make the change of -1 this function: 
https://github.com/NVIDIA-Omniverse/orbit/blob/7a967894946cf6cf52040c8def90892d04856eea/source/extensions/omni.isaac.orbit/omni/isaac/orbit/utils/math.py#L659C5-L659C62
can be more generalized by changing the dim=-1. By doing so, you can accept (...,4) instead of (N,4)

AutonomousHansen, 2024-02-29:
I believe the expected behavior is to handle arguments of the shape (N,4) and (N,3) rather than (...,4) and (...,3). I will make a PR to fix the docstring accordingly.

AutonomousHansen, 2024-03-04:
Thanks @samibouziri for noticing this oversight! I have opened a solving PR upstream.

hhansen-bdai, 2024-03-07:
Thanks for your issue, @samibouziri! Closing as it is solved with upstream #441, and will be released in 0.3.

closed 2024-03-07 by Mayankm96