isaac-sim/IsaacLab #253: AttributeError: 'FrankaCubeLiftEnvCfg' object has no attribute 'control' and ValueError: No checkpoints in the directory: '/home/xxx/xx/xx/2024-02-22_20-25-05' match '/PATH/TO/model.pth'

overrize, 2024-02-22:
2024-02-22 12:35:06 [13,419ms] [Error] [__main__] No checkpoints in the directory: '/home/xxx/xxx/isaacsim/orbit/logs/rsl_rl/franka_reach/2024-02-22_20-25-05' match '/PATH/TO/model.pth'.
2024-02-22 12:35:06 [13,426ms] [Error] [__main__] Traceback (most recent call last):
  File "/home/xxx/xxx/isaacsim/orbit/source/standalone/workflows/rsl_rl/play.py", line 105, in <module>
    main()
  File "/home/xxx/xxx/isaacsim/orbit/source/standalone/workflows/rsl_rl/play.py", line 72, in main
    resume_path = get_checkpoint_path(log_root_path, agent_cfg.load_run, agent_cfg.load_checkpoint)
  File "/home/xxx/xxx/isaacsim/orbit/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/utils/parse_cfg.py", line 198, in get_checkpoint_path
    raise ValueError(f"No checkpoints in the directory: '{run_path}' match '{checkpoint}'.")
ValueError: No checkpoints in the directory: '/home/xxx/xxx/isaacsim/orbit/logs/rsl_rl/franka_reach/2024-02-22_20-25-05' match '/PATH/TO/model.pth'.
I met this trouble while I try to run the example scripts
./orbit.sh -p source/standalone//workflows/robomimic/play.py --task Isaac-Lift-Cube-Franka-IK-Rel-v0 --checkpoint /PATH/TO/model.pth
It does works normally while training,but when I try to run this play.py above,it show what I pased.
I running with conda enviroments.

Agnel-Wang, 2024-02-23:
#### Q1: AttributeError
This is still a bug. [#219]
#### Q2: No checkpoints
Documentation is not the same as code. It does get a bit confusing. 
But looking at the [code](https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/utils/parse_cfg.py) you should use

```bash
./orbit.sh -p source/standalone/workflows/rsl_rl/play.py --task Isaac-Reach-Franka-v0 --num_envs 32 --checkpoint model_999
```

overrize, 2024-02-23:
Thanks a lot！It works smoothly.

overrize, 2024-02-23:
> #### Q1: AttributeError
> 
> This is still a bug. [#219]
> #### Q2: No checkpoints
> 
> Documentation is not the same as code. It does get a bit confusing. But looking at the [code](https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/utils/parse_cfg.py) you should use
> 
> ```shell
> ./orbit.sh -p source/standalone/workflows/rsl_rl/play.py --task Isaac-Reach-Franka-v0 --num_envs 32 --checkpoint model_999
> ```

Thanks a lot！It works smoothly.

closed 2024-10-02 by Dhoeller19