isaac-sim/IsaacLab #257: [Question]

nowionlyseedaylight, 2024-02-27:
### Question

Basic questions, related to robot learning, that are not bugs or feature requests will be closed without reply, because GitHub issues are not an appropriate venue for these.

Advanced/nontrivial questions, especially in areas where documentation is lacking, are very much welcome.

For questions that are related to running and understanding Isaac Sim, please post them at the official [Isaac Sim forums](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/isaac_sim_forums.html).

Mayankm96, 2024-02-29:
The files in Python modules are not directly runnable since they depend on Omniverse to be running first. This is because Omniverse extensions are hot-reloaded when the application is launched. They aren't available directly as Python modules.

There are ideas on how to make this possible but right now they are in the works (which may or may not become a thing). To verify your configs, you will first need to launch the simulation application.

Please check here for more info: https://isaac-orbit.github.io/orbit/source/setup/developer.html#standalone-applications

nowionlyseedaylight, 2024-03-04:
Thank you. I understood how it works. Probably I should figure out other ways to automatize the whole process.

nowionlyseedaylight, 2024-03-05:
@Mayankm96  Thanks for your response.
However, I'm pretty sure it's runnable in python modules (through ./orbit.sh -p ....), since it worked with the last version of orbit.
My colleague made the additional folder in the _orbit/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/manipulation_ to implement the customized code. And it was possible to run the code file via local terminal.
But now, with this new version (or I'm not sure whether it's the problem of my situation) I continually get the error message like above. Is there any possibility for the reason of that error to be the orbit installation process? (for example, the module like omni.client may have been omitted)

pengxiangzang, 2024-09-16:
I'm also confused about this problem. How did you solve it? Could you please give me some advice?

Mayankm96, 2024-09-16:
Are you able to run the demos mentioned here?

> https://isaac-sim.github.io/IsaacLab/source/setup/sample.html#showroom

pengxiangzang, 2024-09-17:
I can run the demos in standalone , but when it comes to franka.py in omni.isaac.orbit_assets of extensions, this problem comes up. So the problem is that I didn't understand the structure of the codes.Thank you for your reply and sorry to take up your time.I'm just starting to learn how to use the isaac lab. Did you follow some video tutorials? The official documentation really confuses me.