isaac-sim/IsaacLab #259: spacemouse

Nimingez, 2024-03-01:
### Question
Have you ever used a space mouse to operate a robotic arm? It seems like there is some issue with HIDAPI.
my hid=1.0.6 hidapi=0.14.0
I just run
 ./orbit.sh -p source/standalone/environments/teleoperation/teleop_se3_agent.py --task Isaac-Lift-Cube-Franka-IK-Rel-v0 --num_envs 1 --device spacemouse
I meet
Traceback (most recent call last):
  File "/home/nmz/Orbit/source/standalone/environments/teleoperation/teleop_se3_agent.py", line 42, in <module>
    from omni.isaac.orbit.devices import Se3Gamepad, Se3Keyboard, Se3SpaceMouse
  File "/home/nmz/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/devices/__init__.py", line 24, in <module>
    from .spacemouse import Se2SpaceMouse, Se3SpaceMouse
  File "/home/nmz/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/devices/spacemouse/__init__.py", line 8, in <module>
    from .se2_spacemouse import Se2SpaceMouse
  File "/home/nmz/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/devices/spacemouse/se2_spacemouse.py", line 10, in <module>
    import hid
  File "/home/nmz/miniforge3/envs/orbit/lib/python3.10/site-packages/hid/__init__.py", line 125, in <module>
    hidapi.hid_get_input_report.argtypes = [ctypes.c_void_p, ctypes.c_char_p, ctypes.c_size_t]
  File "/home/nmz/miniforge3/envs/orbit/lib/python3.10/ctypes/__init__.py", line 387, in ____getattr__
    func = self.__getitem__(name)
  File "/home/nmz/miniforge3/envs/orbit/lib/python3.10/ctypes/__init__.py", line 392, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /lib/x86_64-linux-gnu/libhidapi-hidraw.so: undefined symbol: hid_get_input_report

Mayankm96, 2024-03-01:
Can you try the solution here: https://github.com/Poohl/joycontrol/issues/17#issuecomment-1253062321

Nimingez, 2024-03-04:
hidapi project does not handle the Python wrapper. I can not solve the problem

closed 2024-03-06 by Nimingez