isaac-sim/IsaacLab #261: [Question] Quadcopter Examples/Guides

mohamedamr13, 2024-03-05:
### Question

Are there any Quadcopter examples/tutorials I could check for Orbit. There's a simple Quadcopter example in OmniverseIsaacSim but in the Orbit paper it is mentioned that Fluid Dynamics are not supported on IsaacSim while they are supported in Orbit, so I wondered what difference it would make in terms of task complexity and sim2real gap.

Mayankm96, 2024-03-06:
Apologies for any misunderstanding here. Fluid is not supported in Isaac Gym but it is supported in Isaac Sim and Omniverse. Orbit relies on Omniverse capabilities for physics simulation and rendering engine.

For quadrotors, the kind of fluid simulation you're expecting is not possible at this point. However, many works have used an approximate model for quadrotors and can do reliable sim-to-real. These include:
* https://github.com/ntnu-arl/aerial_gym_simulator
* https://omnidrones.readthedocs.io/en/latest/
* https://github.com/PegasusSimulator/PegasusSimulator

Since these are frameworks from research labs, it is difficult to comment on whether their developers would go through the engineering effort to transition those environments to Orbit. However, if these are of interest, we definitely would like to support such efforts from our side.

mohamedamr13, 2024-03-08:
@Mayankm96 thanks a lot for the clarification

closed 2024-04-12 by Mayankm96