isaac-sim/IsaacLab #266: [Question] simulation_app.close() hangs

hesic73, 2024-03-07:
I'm setting up Isaac Sim on a new computer, but it seems the application cannot close normally.
### My environment
- Ubuntu 22.04
- NVIDIA GeForce RTX 3060, Driver Version: 550.54.14
- 12th Gen Intel(R) Core(TM) i5-12600K
- Isaac Sim 2023.1.1

### Issue

```python
from omni.isaac.orbit.app import AppLauncher

if __name__ == "__main__":
    app_launcher = AppLauncher(headless=True)
    simulation_app = app_launcher.app
    print(simulation_app.is_running())
    simulation_app.close()
    print(simulation_app.is_running())
```

When I run the script above, I have the following output:
```bash
True
[11.896s] Simulation App Shutting Down
2024-03-07 03:31:40 [11,898ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/admin1/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
X connection to |G�I� broken (explicit kill or server shutdown).
../../../source/framework/PluginImpl.cpp:602:void carb::Plugin::unload()(): Assertion (m_initialized.load() != True) failed.
Trace/breakpoint trap (core dumped)
```
But when I try to import something after the application is launched, it will hang at `simulation_app.close()`:

```python
from omni.isaac.orbit.app import AppLauncher

if __name__ == "__main__":
    app_launcher = AppLauncher(headless=True)
    simulation_app = app_launcher.app
    import omni.isaac.orbit_tasks
    print(simulation_app.is_running())
    simulation_app.close()
    print(simulation_app.is_running())

```

```
True
[12.604s] Simulation App Shutting Down
2024-03-07 03:34:58 [12,607ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/admin1/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
X connection to �X broken (explicit kill or server shutdown).
2024-03-07 03:35:03 [18,038ms] [Warning] [omni.kit.telemetry.plugin] failed to wait for the extension's startup tasks to complete.  This may result in a hang.
```

Then I try this:
```python
from omni.isaac.kit import SimulationApp

if __name__ == "__main__":
    app = SimulationApp({"headless": True})
    print(app.is_running())
    app.close()
    print(app.is_running())

```
**The program still hangs at `app.close()`.**
```
True
[11.893s] Simulation App Shutting Down
2024-03-07 03:38:31 [11,895ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/admin1/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
X connection to m�LB# broken (explicit kill or server shutdown).
2024-03-07 03:38:37 [18,017ms] [Warning] [omni.kit.telemetry.plugin] failed to wait for the extension's startup tasks to complete.  This may result in a hang.
```

When I run the above script in a **clean Isaac Sim conda environment**, the output is a bit different:
```
True
[11.916s] Simulation App Shutting Down
2024-03-07 03:45:00 [11,918ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/admin1/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
XIO:  fatal IO error 9 (Bad file descriptor) on X server "�?"
      after 51 requests (51 known processed) with 0 events remaining.
2024-03-07 03:45:06 [18,042ms] [Warning] [omni.kit.telemetry.plugin] failed to wait for the extension's startup tasks to complete.  This may result in a hang.
```

hesic73, 2024-03-08:
I find that the program will crash at `simulation_app.close()` when `fastShutdown=True`. At least I won't need to kill the process manually.

Dhoeller19, 2024-10-02:
Hi @hesic73, this should be fixed by now. Your repro works for me.

closed 2024-10-02 by Dhoeller19