isaac-sim/IsaacLab #267: Jacobian calculation for quadruped robot

catachiii, 2024-03-07:
Hi @Mayankm96 , I had issues getting the correct Jacobian for quadrupeds from Orbit or Isaac Sim.

I was running the example Isaac-Velocity-Flat-Anymal-C-v0. After the step function, I tried to access jacobian through env.scene.articulations['robot'].root_physx_view.get_jacobians() and by accessing the FL_FOOT vel jacobian wrt joint vel by [0, 4, 0:3, -12:], I found almost all entries are zero, which is wrong...

I'm not sure what happened... this can be very helpful for IK.

Thanks!

hhansen-bdai, 2024-03-07:
Hello @catachiii, thanks for opening this issue! :smile:  However, I think it would be better if you changed this to the Bug Report template. If you fill out the template with the information it requests, we will address this as soon as we can :+1:

catachiii, 2024-03-07:
Sure I can do that 😄

Mayankm96, 2024-03-12:
@catachiii 

How did you obtain the link index for the Jacobians. We identified an issue when you try to work directly with PhysX articulation view. The indexing of body names inside the articulation view is not the same as that in the body view.

For body names in the ArticulationView, we follow the rigid prim view ordering:

```python
    @property
    def body_names(self) -> list[str]:
        """Ordered names of bodies in articulation (through rigid body view)."""
        prim_paths = self.body_physx_view.prim_paths[: self.num_bodies]
        return [path.split("/")[-1] for path in prim_paths]
```

However, in the articulation view, the body ordering is different:

```python
   @property
    def body_names(self) -> list[str]:
        """Ordered names of bodies in articulation (through articulation view)."""
        return self._root_physx_view.shared_metatype.link_names
```

This yields two different listing (for example, for ANYmal):

```
Link names through body view: ['base', 'LF_HIP', 'LF_THIGH', 'LF_SHANK', 'LF_FOOT', 'LH_HIP', 'LH_THIGH', 'LH_SHANK', 'LH_FOOT', 'RF_HIP', 'RF_THIGH', 'RF_SHANK', 'RF_FOOT', 'RH_HIP', 'RH_THIGH', 'RH_SHANK', 'RH_FOOT']
Link names through articulation view: ['base', 'LF_HIP', 'LH_HIP', 'RF_HIP', 'RH_HIP', 'LF_THIGH', 'LH_THIGH', 'RF_THIGH', 'RH_THIGH', 'LF_SHANK', 'LH_SHANK', 'RF_SHANK', 'RH_SHANK', 'LF_FOOT', 'LH_FOOT', 'RF_FOOT', 'RH_FOOT'
```

So in your case, the body indexing needs to be done w.r.t. articulation view as you read the Jacobian quantity from the articulation view. Can you please check this and see if the issue still persists?

catachiii, 2024-03-13:
@Mayankm96 

Thanks for your clarification! Indeed, the index messed things up. 

I used the find_bodies function in rigid_object to find the index...
`    def find_bodies(self, name_keys: str | Sequence[str]) -> tuple[list[int], list[str]]:
        """Find bodies in the articulation based on the name keys.

        Please check the :meth:`orbit.utils.string_utils.resolve_matching_names` function for more
        information on the name matching.

        Args:
            name_keys: A regular expression or a list of regular expressions to match the body names.

        Returns:
            A tuple of lists containing the body indices and names.
        """
        return string_utils.resolve_matching_names(name_keys, self.body_names)`

This from my side will return the indices in rigid body view.

Mayankm96, 2024-03-13:
Yes the data in Articulation follows rigid body view ordering. Only data from PhysX articulation view for links are in a different ordering. I have an MR under review right now which makes all this in the same indexing. It should get merged later today or this week.

Mayankm96, 2024-03-15:
Merged the MR that should fix the issue: e3c40acf5e6433c1d88b54a6e9d45bd6d3e73152

Please reopen if the problem still persists.

closed 2024-03-15 by Mayankm96