isaac-sim/IsaacLab #268: [Bug Report] WebRTC streaming does not work on Orbit scripts with Isaac Sim 2023.1.1

ozhanozen, 2024-03-07:
If I use the 2023.1.1 version of Isaac Sim as a base in Orbit and enable WebRTC livestream on Orbit scripts, WebRTC does not start correctly, and I get the following errors on the terminal:

```
[ext: omni.kit.livestream.webrtc-2.3.2] startup
[Error] [carb.livestream.plugin] Stream Server: Net Stream Creation failed, 0x800B1000
[Error] [carb.livestream.plugin] Could not initialize streaming components
[Error] [carb.livestream.plugin] Couldn’t initialize the capture device.
[ext: omni.services.streamclient.webrtc-1.3.8] startup
```

### Steps to reproduce

Run random_ageny.py with WebRTC enabled (livestream=3) on any environment. 

### System Info

- Commit: [93f3eff55341c2a07ca9e3f15717622d24d94584]
- Isaac Sim Version: [2023.1.1]
- OS: [Ubuntu 20.04]
- GPU: [RTX A6000]
- CUDA: [12.0]
- GPU Driver: [525.60.11]

### Additional context

WebRTC works correctly with Orbit scripts if the Isaac Sim version is 2023.1.0-hotfix.1. The problem is specific to the 2023.1.1 version.

Interestingly, even on version 2023.1.1, running directly `runheadless.webrtc.sh` from the container successfully enables the WebRTC. Hence, the problem is most probably related to Orbit.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] WebRTC livestream works with orbit scripts

ozhanozen, 2024-04-22:
Update: it seems like the problem is not specific to Orbit, as the livestream.py example of Isaac Sim also fails the same way when WebRTC is enabled. 

I wonder if there is a workaround we could follow from Orbit.

Mayankm96, 2024-04-26:
Closing this issue to continue the discussion in #381.

closed 2024-04-26 by Mayankm96