isaac-sim/IsaacLab #270: [Question] Anymal actuator network joint order

Vassil17, 2024-03-07:
Hi,

I noticed that the joint_names ordering for the Anymal C is ['LF_HAA', 'LH_HAA', 'RF_HAA', 'RH_HAA', 'LF_HFE', 'LH_HFE', 'RF_HFE', 'RH_HFE', 'LF_KFE', 'LH_KFE', 'RF_KFE', 'RH_KFE'].

In the isaac gym environments the joints are defined as ['LF_HAA', 'LF_HFE', 'LF_KFE', 'LH_HAA', 'LH_HFE', 'LH_KFE', 'RF_HAA', 'RF_HFE', 'RF_KFE', 'RH_HAA', 'RH_HFE', 'RH_KFE'].

Which order does the actuator network model use? Both codebases seem to just pass the joints in their respective order, while using the same actuator network. Are the joints being reordered at any point which I have missed?

Mayankm96, 2024-03-07:
No, they're not. The actuator model is modeling each actuator, i.e. it does not depend on how the other joints/actuators behave (basically a SISO or decoupled model).

Vassil17, 2024-03-08:
Fantastic, thanks for clarifying!

closed 2024-03-08 by Vassil17