isaac-sim/IsaacLab #273: Using Action Graphs for agent Actions

MaltheT, 2024-03-08:
### Discussed in https://github.com/NVIDIA-Omniverse/orbit/discussions/260

<div type='discussions-op-text'>

<sup>Originally posted by **MaltheT** March  5, 2024</sup>
Hi,

I have a setup where I use an Action Graph with the Ackermann Steering node to drive my robot. However, it doesn't articulate the joints in my orbit sim when I use a preconfigured 3.0 speed. I have tested that it works in regular Isaac Sim. How does orbit work with Action Graphs, and can I use my Action Graph's interface for the Actions of my RL agent (i.e. speed and steeringAngle fields)?

[image]
</div>

MaltheT, 2024-03-08:
## More detailed question
I'm curious if using this ackermann_steering node to define my action space in orbit is possible. Right now I have a setup in orbit like this:
```python
F1TENTH_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path=f"{F1TNEHT_PROJET_DIR}/robot/instanceable/f1tenth.usd",
    ),
    init_state=ArticulationCfg.InitialStateCfg(
        pos=(0.0, 0.0, 1.0)
    ),
    actuators={
        "wheels": ImplicitActuatorCfg(
            joint_names_expr=['wheel_.*'],
            effort_limit=None,
            velocity_limit=None,
            stiffness=None,
            damping=None
        ),
        "rotators": ImplicitActuatorCfg(
            joint_names_expr=["rotator_.*"],
            effort_limit=None,
            velocity_limit=None,
            stiffness=None,
            damping=None
        ),
    },
)
```
And for the actions:
```python
@configclass
class ActionsCfg:
    """Action specifications for the MDP."""
    steering = mdp.JointPositionActionCfg(asset_name="robot", joint_names=["rotator_left", "rotator_right"], scale=1.0)
    drive = mdp.JointVelocityActionCfg(asset_name="robot", joint_names=["wheel_back_left", "wheel_back_right", "wheel_front_left", "wheel_front_right"], scale=1.0)
```
However, this means that my agent thinks that it can control each joint independently, which is not the case for the physical system. Previously in my simulation, I used the ackermann_steering node to solve this issue, to implement the correct control behaviour. All of the examples I have looked at in the orbit repo perform control directly on the joints. However, I'm curious if there is a way to define the action space using the action graphs as they are used to control a lot of wheeled robots in Isaac Sim, i.e. something like this:
```python
@configclass
class ActionsCfg:
    """Action specifications for the MDP."""
    # example:
    ackermann = mdp.OmniGraphCfg(asset_name="robot", node_name=["ackermann_drive"], scale=1.0) #(This code does not work)
```
I know that the code I have provided does not work, but I hope that the desired behavior is clear. Is there a way to achieve this through the current actions or commands APIs? Any help is highly appreciated!

Mayankm96, 2024-03-12:
Closing this one here to just keep in #260

closed 2024-03-12 by Mayankm96