isaac-sim/IsaacLab #276: [Bug Report]

Vassil17, 2024-03-08:
### Describe the bug

Training the Anymal environments crashes when using randomising the physics material.

### Steps to reproduce

Running any training script on the flat or rough anymal envs. For example:
```
./orbit.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Velocity-Flat-Anymal-C-v0 --headless
```

# Error Log:
```
2024-03-08 15:59:39 [84,519ms] [Error] [__main__] Traceback (most recent call last):
  File "/home/user/miniconda3/envs/my_env/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/user/test/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/rl_task_env.py", line 99, in __init__
    self.randomization_manager.randomize(mode="startup")
  File "/home/user/test/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/managers/randomization_manager.py", line 160, in randomize
    term_cfg.func(self._env, env_ids, **term_cfg.params)
  File "/home/user/test/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/mdp/randomizations.py", line 74, in randomize_rigid_body_material
    indices = torch.tensor(asset_cfg.body_ids, dtype=torch.int).repeat(len(env_ids), 1)
TypeError: 'slice' object cannot be interpreted as an integer

```

### System Info

Describe the characteristic of your environment:

- Commit: 93f3eff
- Isaac Sim Version: 2023.1.1-rc.8+2023.1.688.573e0291.tc
- OS: Ubuntu 20.04
- GPU: RTX 3060
- CUDA: 12.4
- GPU Driver: 550.40.07

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Allow physics materials to be randomized again.

Mayankm96, 2024-03-09:
Thanks for pointing this out. The fix is under review. Hope to have it merged soon. As a hot-fix, you can comment out these two lines of code:

https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/managers/scene_entity_cfg.py#L140-L142

Mayankm96, 2024-03-12:
This should be fixed now with [e444df7](https://github.com/isaac-orbit/orbit/commit/e444df7d75094dfd1339e9bfea7e2ac24448c4f4)

Thanks for reporting!

weifeng-lt, 2024-05-23:
Where is [e444df7](https://github.com/isaac-orbit/orbit/commit/e444df7d75094dfd1339e9bfea7e2ac24448c4f4)?
The bug still exists in branch 'main' !