isaac-sim/IsaacLab #277: [Bug Report] Bug title

KyleM73, 2024-03-08:
### Describe the bug

The default values for many values in `source/extensions/omni.isaac.orbit/omni/isaac/orbit/managers/scene_entity_cfg.py` where changed from `None` to `slice(None)` in commit d682c8d. This creates an error in `randomize_rigid_body_material` in `orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/mdp/randomizations.py"` because ` torch.tensor(asset_cfg.body_ids, dtype=torch.int)` expects asset_cfg.body_ids to contain ints but it contains a slice, raising the exception below. 

### Steps to reproduce

`orbit -p orbit/source/standalone/workflows/rsl_rl/train.py --headless --task Isaac-Velocity-Flat-Unitree-Go1-v0`

(I verified that this bug occurs with multiple of the default envs)

```
Traceback (most recent call last):
  File "/isaac-sim/kit/python/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/workspace/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/rl_task_env.py", line 99, in __init__
    self.randomization_manager.randomize(mode="startup")
  File "/workspace/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/managers/randomization_manager.py", line 160, in randomize
    term_cfg.func(self._env, env_ids, **term_cfg.params)
  File "/workspace/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/mdp/randomizations.py", line 74, in randomize_rigid_body_material
    indices = torch.tensor(asset_cfg.body_ids, dtype=torch.int).repeat(len(env_ids), 1)
TypeError: 'slice' object cannot be interpreted as an integer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/workspace/orbit/source/standalone/workflows/rsl_rl/train.py", line 142, in <module>
    main()
  File "/workspace/orbit/source/standalone/workflows/rsl_rl/train.py", line 96, in main
    env = gym.make(args_cli.task, cfg=env_cfg, render_mode="rgb_array" if args_cli.video else None)
  File "/isaac-sim/kit/python/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 814, in make
    raise type(e)(
TypeError: 'slice' object cannot be interpreted as an integer was raised from the environment creator for Isaac-Velocity-Flat-Unitree-Go1-v0 with kwargs 
```

### System Info

Describe the characteristic of your environment:

- Commit: main, d682c8d
- Isaac Sim Version: 2023.1.0.hotfix
- OS: Ubuntu 20.04
- GPU: RTX 3080

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Either revert default entity cfg values to NoneType or adapt mdp functions to accept slice inputs (or convert slice to ints in mdp function calls like what is done for empty env_ids)

KyleM73, 2024-03-08:
I realized this is the same problem referenced in issue #276

Mayankm96, 2024-03-09:
Thanks for pointing this out. The fix is under review. Hope to have it merged soon. As a hot-fix, you can comment out these two lines of code:

https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/managers/scene_entity_cfg.py#L140-L142

closed 2024-03-12 by Mayankm96