isaac-sim/IsaacLab #282: [Proposal] Don't overwrite height field sub-terrains parameters in TerrainGenerator

Wojcik98, 2024-03-13:
### Proposal

Currently, `TerrainGenerator` class overwrites configs of height field sub-terrains with its own parameters: [code](https://github.com/NVIDIA-Omniverse/orbit/blob/8dea21a8ac805ab4dd804fc9927110f3c27f8b3b/source/extensions/omni.isaac.orbit/omni/isaac/orbit/terrains/terrain_generator.py#L95). I propose to remove this part and those settings from [TerrainGeneratorCfg](https://github.com/NVIDIA-Omniverse/orbit/blob/8dea21a8ac805ab4dd804fc9927110f3c27f8b3b/source/extensions/omni.isaac.orbit/omni/isaac/orbit/terrains/terrain_generator_cfg.py#L144).

As far as I can tell those settings in `TerrainGeneratorCfg` are used only for overwriting, so it wouldn't cause direct problems in other parts. One problem it may cause is break compatibility slightly, as people would need to remove those parameters when creating `TerrainGeneratorCfg`.

I can provide PR if the discussion agrees to this proposal.

### Motivation

Currently, it is impossible to set different settings for each sub-terrain, as they will be overwritten by `TerrainGenerator`. It directly affected me, because I needed to increase accuracy of only one sub-terrain but leave the rest with default values for optimization.

Additionally, I feel like `TerrainGenerator` simply shouldn't be responsible for changing those settings, only for reading them. It is confusing that it can change them. It also adds that much complexity to the class which is not necessary, it can be handled by `HfTerrainBaseCfg` alone. It is also specific to this one subclass of `SubTerrainBaseCfg` and changes its internal behavior.

Kinda similar argument could be made for overwriting `size` field but it less obvious how to handle differently sized sub-terrains and whether it should be possible, so I would currently leave it out of this discussion.

### Alternatives

Alternative would be to overwrite only if the values were not explicitly filled by the user. This might be doable for `horizontal_scale` and `vertical_scale` as they could have assigned `None` as default value, but for `slope_threshold` `None` is a valid value so it would be difficult to detect whether user didn't fill it or if they actually meant `None`. If there is another way to accomplish it, it would provide better backward compatibility, though I really think this functionality simply shouldn't be in this class.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

- [ ] It is possible to assign different settings to each sub-terrain passed to `TerrainGenerator`.

Mayankm96, 2024-03-13:
Hi @Wojcik98 ,

Thank you for bringing this up. I see your point that this overwriting in the code makes things less transparent.

However, to understand your use-case better for having different sizes of sub-terrains. This would require reworking some of the internal workings on arranging the terrains on a grid. Right now, we align them easily because we know they all share the same size.

If sub-terrains have different sizes, then defining a grid is no longer "clean." How do you propose to tackle this? Or is this something you don't care about for your application?

Mayankm96, 2024-03-13:
Hi @Wojcik98 ,

I think you are right about this. 

We set certain [default values for the height field](https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/terrains/terrain_generator_cfg.py#L144-L162) in the `SubTerrainGeneratorCfg`, so it is confusing as to why we don't just set these in the [base config for height field](https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/terrains/height_field/hf_terrains_cfg.py#L26-L32). If I recall correctly, the motivation to set them from `SubTerrainGeneratorCfg` was to change the configuration for all of them at once (since usually that's what many would do, unless they want to tune this for each sub-terrain).

We can be compatible with the way you suggest. Have `None` or `-1` values by default in `SubTerrainGeneratorCfg`, and if users are setting these from `SubTerrainGeneratorCfg`, we:
1. throw a deprecation warning and notify them of a coming breaking change so they can adapt
2. keep it like this and support both your and general use cases

 @Dhoeller19  @nikitardn what do you think?

dmtrung14, 2024-12-30:
Following up on this conversation, I wonder if the issue affects mesh terrains? It seems to me that this chunk of codes in `terrain_generator.py` only affects hight fields terrains:
```python
for sub_cfg in self.cfg.sub_terrains.values():
    # size of all terrains
    sub_cfg.size = self.cfg.size
    # params for height field terrains
    if isinstance(sub_cfg, HfTerrainBaseCfg):
        sub_cfg.horizontal_scale = self.cfg.horizontal_scale
        sub_cfg.vertical_scale = self.cfg.vertical_scale
        sub_cfg.slope_threshold = self.cfg.slope_threshold
```
but this discussion seems to be referencing `SubTerrainGeneratorCfg` as a whole which, afaik, `MeshTerrainGeneratorCfg` also inherits from.

closed 2026-09-30 by ooctipus