isaac-sim/IsaacLab #284: [Question] Slow Loading with Simulation App

Aronnaxx, 2024-03-14:
Hi all,

I’ve been experimenting with Orbit and getting it to run on Windows with three separate environments:

    Asus 2022 M16 laptop (3070ti + Intel 12700k CPU, and 40GB DDR5 4500, installed on NVME) running Windows 11
    Desktop PC (4090 + 7800x3D, and 64gb DDR5 5000 RAM, installed on NVME) running Windows 11
    Desktop PC (same as above) with Ubuntu 22.04

When I run Isaac sim on the laptop and on the Linux install, it’s incredibly fast. The laptop running Windows 11 will begin the simulation and launch the app within 30 seconds of starting it, and the linux will do so in less than 20 seconds.

However, when I attempt to run the simulation on Windows natively, it can take upwards of 3 minutes to begin the simulation (longer if I attempt to run the sim via python and not just the Omniverse UI)

I’ve tried replicating the laptop’s settings, checked that hardware acceleration is on, and attempted multiple versions of Isaac Sim on the desktop but there is no improvement. I’ve also confirmed that both the desktop and laptop are on the latest versions of their respective drivers, but I still haven’t seen any performance / loading improvements on the desktop. I even attempted a fresh install of Windows 11 on the desktop, but it still did not help the loading times.

Is there a resolution to this, or a way to speed up the loading? It seems like the desktop running Windows 11 would be FASTER than the laptop, especially considering the difference in hardware specifications.

These were all run using the Orbit framework’s create_scene.py from their tutorial

Logs from the laptop:
[kit_20240312_174503.log](https://forums.developer.nvidia.com/uploads/short-url/3Uj3nmkZ76JeRmhUI85FPVefcMd.log) (2.2 MB)

Total load time for the scene: 25 seconds (start to finish). Subsequent runs took around ~15 seconds.

Desktop Linux logs:
[kit_20240312_180238.log](https://forums.developer.nvidia.com/uploads/short-url/j9qBVI2IUZ44ajoaVrctwWfSCJr.log) (2.2 MB)

Total load time (including scene), 15 seconds, subsequent runs were ~8 seconds

Desktop Windows logs
[kit_20240312_174944.log](https://forums.developer.nvidia.com/uploads/short-url/mKgaW09YGxGYhNna6Xp9lJWGj4K.log) (2.2 MB)

Total load time for the scene: 244 seconds (almost 10x slower). Subsequent runs were ~125 seconds.

I’ve noticed that this is where it hangs specifically, although I’m not sure how to expand the logging. Essentially, it seems to get stuck on the simulation app starting.

```
[5.329s] [ext: omni.importer.urdf-1.6.1] startup
[5.371s] [ext: omni.importer.mjcf-1.1.0] startup
[5.378s] [ext: omni.kit.window.stats-0.1.4] startup
[5.380s] [ext: omni.isaac.sim.python-2023.1.1] startup
[5.381s] Simulation App Starting
2024-03-13 00:55:35 [5,359ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/ecoMode/enabled has been requested but is not available
2024-03-13 00:55:35 [5,360ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/directLighting/enabled has been requested but is not available
2024-03-13 00:55:35 [5,360ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/reflections/enabled has been requested but is not available
2024-03-13 00:55:35 [5,360ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/translucency/enabled has been requested but is not available
2024-03-13 00:55:35 [5,361ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/raytracing/subsurface/enabled has been requested but is not available  
2024-03-13 00:55:35 [5,361ms] [Warning] [omni.kit.widget.settings.settings_model] a value for setting /rtx/caustics/enabled has been requested but is not available
2024-03-13 00:55:37 [7,163ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2024-03-13 00:55:37 [7,163ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[124.803s] app ready
2024-03-13 00:57:35 [125,149ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2024-03-13 00:57:35 [125,149ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[125.378s] RTX ready
[125.836s] Simulation App Startup Complete
[126.837s] [ext: omni.physx.fabric-105.1.12-5.1] startup
```

2361098148, 2024-03-14:
I got the same questions, is there any body could help?

Aronnaxx, 2024-03-15:
I'm not certain, but checking the resource monitor shows that Python isn't using as much of the CPU as it should on the computer when booted into linux versus windows?

[image]

Dhoeller19, 2024-10-02:
Dear @Aronnaxx, the simulation starting times have been improved with the latest releases of Isaac Lab.
But yes, there can be differences between linux and windows.