isaac-sim/IsaacLab #290: [Bug Report] Bug title

Felix-Zhenghao, 2024-03-15:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Unable to run the Isaac-Velocity-Rough-Unitree-Go1-v0 envirionment.

### Steps to reproduce
Just use the following command (add --headless will produce the same error message):

```
./orbit.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Velocity-Rough-Unitree-Go1-v0 --num_envs 64
```

### System Info

Describe the characteristic of your environment:

- Commit: The newest commit
- Isaac Sim Version: 2023.1.1
- OS: Ubuntu 20.04
- GPU: RTX 3080
- CUDA: 12.2
- GPU Driver: 535.161.07

### Additional context

Running Isaac-Cartpole-v0 is OK. 

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

Mayankm96, 2024-03-15:
Is this the error you are seeing as well?

```
2024-03-15 09:03:44 [2,890ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[omni.isaac.sim.python-2023.1.1] dependency: (name: 'omni.kit.streamsdk.plugins', tag: '', version: '2.5.2', exact: 1, optional: 0) can't be satisfied. Available versions:
	 - [omni.kit.streamsdk.plugins-2.6.2+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.5.1+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.5.0+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.3.7+105.1.lx64.r] (registry)
	 - [omni.kit.streamsdk.plugins-2.0.3+105.0.lx64.r] (registry)

2024-03-15 09:03:44 [2,890ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
Traceback (most recent call last):
  File "/home/mayank/git_nv/branches/orbit_release/source/standalone/workflows/rsl_rl/train.py", line 47, in <module>
    app_launcher = AppLauncher(args_cli, experience=app_experience)
  File "/home/mayank/git_nv/branches/orbit_release/source/extensions/omni.isaac.orbit/omni/isaac/orbit/app.py", line 109, in __init__
    self._create_app()
  File "/home/mayank/git_nv/branches/orbit_release/source/extensions/omni.isaac.orbit/omni/isaac/orbit/app.py", line 435, in _create_app
    self._app = SimulationApp(self._simulationapp_config, experience=self._simulationapp_experience)
  File "/home/mayank/git_nv/branches/orbit_release/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 195, in __init__
    from .utils import create_new_stage, open_stage, set_livesync_stage
  File "/home/mayank/git_nv/branches/orbit_release/_isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/utils.py", line 17, in <module>
    from omni.kit.usd import layers
ModuleNotFoundError: No module named 'omni.kit.usd'

```

Felix-Zhenghao, 2024-03-16:
This is the error I saw. But both errors seem to have relationship with the 'kit' stuff? Can that be solved?

```
2024-03-15 07:35:39 [30,547ms] [Warning] [omni.client.python] Detected a blocking function. This will cause hitches or hangs in the UI. Please switch to the async version:
  File "/home/rezwan/workspace_felix/tom_and_jerry/run_env.py", line 76, in <module>
  File "/home/rezwan/workspace_felix/tom_and_jerry/run_env.py", line 45, in main
  File "/home/rezwan/workspace_felix/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/rl_task_env.py", line 85, in __init__
  File "/home/rezwan/workspace_felix/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/base_env.py", line 135, in __init__
  File "/home/rezwan/workspace_felix/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/simulation_context.py", line 369, in reset
  File "/home/rezwan/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 624, in reset
  File "/home/rezwan/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 596, in initialize_physics
  File "/home/rezwan/.local/share/ov/pkg/isaac_sim-2023.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 701, in step
  File "/home/rezwan/workspace_felix/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/asset_base.py", line 92, in <lambda>
  File "/home/rezwan/workspace_felix/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/asset_base.py", line 255, in _initialize_callback
  File "/home/rezwan/workspace_felix/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py", line 518, in _initialize_impl
  File "/home/rezwan/workspace_felix/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py", line 615, in _process_actuators_cfg
  File "/home/rezwan/workspace_felix/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/actuators/actuator_net.py", line 129, in __init__
  File "/home/rezwan/workspace_felix/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/utils/assets.py", line 133, in read_file
  File "/home/rezwan/.local/share/ov/pkg/isaac_sim-2023.1.1/kit/extscore/omni.client/omni/client/__init__.py", line 491, in read_file
```

Mayankm96, 2024-03-16:
Please make sure you can connect to Nucleus: https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_faq.html#assets-and-nucleus

zoctipus, 2024-04-02:
> Is this the error you are seeing as well?
>
> ```
> 2024-03-15 09:03:44 [2,890ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
> 	[omni.isaac.sim.python-2023.1.1] [...]

I am seeing this issue after I pulled the update today, I made sure all services in nucleus setting http://localhost:3080/ is on, and check `~/.local/share/ov/data/Kit/Isaac-Sim/2023.1/user.config.json` has the same configuration as [](https://docs.omniverse.nvidia.com/isaacsim/latest/installation/install_faq.html#assets-and-nucleus). But this issue is still there.

closed 2024-10-04 by Dhoeller19