isaac-sim/IsaacLab #292: [Bug Report] ` --checkpoint` broken

VladimirFokow, 2024-03-15:
### Describe the bug

Here: https://isaac-orbit.github.io/orbit/source/setup/sample.html#reinforcement-learning

In this example:
```bash
./orbit.sh -p source/standalone/workflows/rsl_rl/play.py --task Isaac-Reach-Franka-v0 --num_envs 32 --checkpoint /PATH/TO/model.pth
```

we are supposed to provide `/PATH/TO/model.pth`

- First of all, in my logs, the models with `.pt` extension are saved, not `.pth`

- Secondly, it would be nice to write in the docs how to find this path, so that there's no need for the user to go looking for the logic into the code.

- Thirdly, most importantly, it's completely broken even if you provide the path:

```bash
---------------------------------
Traceback (most recent call last):
  File "/home/fokow/work/Orbit/source/standalone/workflows/rsl_rl/play.py", line 106, in <module>
    main()
  File "/home/fokow/work/Orbit/source/standalone/workflows/rsl_rl/play.py", line 74, in main
    resume_path = get_checkpoint_path(log_root_path, agent_cfg.load_run, agent_cfg.load_checkpoint)
  File "/home/fokow/work/Orbit/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/utils/parse_cfg.py", line 215, in get_checkpoint_path
    raise ValueError(f"No checkpoints in the directory: '{run_path}' match '{checkpoint}'.")
ValueError: No checkpoints in the directory: '/home/fokow/work/Orbit/logs/rsl_rl/franka_reach/2024-03-15_21-08-21' match '/home/fokow/work/Orbit/logs/rsl_rl/franka_reach/2024-03-15_21-08-21'.
```

In [this line](https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/utils/parse_cfg.py#L206) my `checkpoint` equals `run_path` for some reason, that's why no matches are found.

VladimirFokow, 2024-03-16:
What should the logic be: 
- the latest checkpoint automatically selected, 
- or exactly the one specified by the user?

the code attempts to do both 😑 and achieves none.

---

If the latest checkpoint automatically selected - I suggest using the [`natsort`](https://github.com/SethMMorton/natsort) library instead of `.sort(key=lambda m: f"{m:0>15}")`  [here](https://github.com/NVIDIA-Omniverse/orbit/blob/475b3f791490fa9ac5d5b593eafe66c933c8924f/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/utils/parse_cfg.py#L211C22-L211C54)

Mayankm96, 2024-03-19:
1. You are right about pth. That is an error.
2. You need to specify the "load_run" and the "checkpoint" file (as the args say). The paths are resolved automatically

```bash
./orbit.sh -p source/standalone/workflows/rsl_rl/play.py --task Isaac-Rough-Anymal-C-Play-v0 --load_run 2024-03-11_16-11-38 --checkpoint model_300.pt
```

I agree. This is not very intuitive. Feel free to send an MR with the required clarifications :)

closed 2024-10-04 by Dhoeller19