isaac-sim/IsaacLab #301: [Proposal] Get Analytical Jacobian

catachiii, 2024-03-21:
### Proposal

Calculate the analytical Jacobian (body pos/vel w.r.t generalized coordinates) without calling the simulation step function (with the imported URDF/USD).

### Motivation

Analytical Jacobian is important for a wider range of robotic controls since they are fast and accurate. The method mentioned in [Joonho's Science Robotics paper ](https://www.science.org/doi/pdf/10.1126/scirobotics.abc5986) requires IK to work. The current estimation of the Jacobian is not accurate, and it is difficult to have exactly the same implementation on the C++ side for sim2real transfer. 

### Alternatives

There are also other libraries like Pinocchio trying to calculate the analytical Jacobian, but they need to be parallelized and could not be easily tested with Isaac Sim. 

### Additional context

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Output parallelized analytical Jacobian in either data to from Articulation in Isaac Sim.

Mayankm96, 2024-03-22:
We don't want to expose the Jacobians through the `data` field as this is an expensive quantity to fetch due to its size. You can use the `root_physx_view` to query from the Jacobians.

However, I think your question is more about "updating" these kinematic quantities without doing PhysX step. Unfortunately, this isn't possible. You can use Warp to make a sim model to query Jacobians. Please check their documentation on how to do so.

catachiii, 2024-03-25:
Thanks @Mayankm96, I'll take a look!

closed 2024-04-12 by Mayankm96