isaac-sim/IsaacLab #306: [Proposal] don't repeat `cfg`

VladimirFokow, 2024-03-21:
### Proposal

https://github.com/NVIDIA-Omniverse/orbit/blob/83d62e212dde22ebc80f140388417fbca12229ed/docs/source/tutorials/00_sim/spawn_prims.rst?plain=1#L67-L68

Repeating the `cfg` inside of the brackets goes against the [DRY principle](https://en.wikipedia.org/wiki/Don%27t_repeat_yourself)

`cfg` is already the object that's calling `func` -> so `func` can automatically know about it - no need to make the user repeat it.

it's possible using [partials](https://stackoverflow.com/a/15331841) (if there's no better way, e.g. larger code redesign):

<br />

### Motivation

Better code design.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Either:

- [ ] fixed
- [ ] explained in the docs why this way

VladimirFokow, 2024-03-21:
plus, the `func` isn't really a descriptive method name. Maybe `spawn` would be better?

Dhoeller19, 2024-10-04:
@VladimirFokow thanks for the suggestion, we are working on it.

StafaH, 2026-09-30:
Hi @VladimirFokow, thanks for the suggestion. We have begun to implement a functional style in: https://github.com/isaac-sim/IsaacLab/pull/8085, you can now use `object = instantiate(cfg)`.

closed 2026-09-30 by StafaH