isaac-sim/IsaacLab #322: [Question] Difference between tactile sensor and support contact sensor

ruoshiwen, 2024-03-23:
### Tactile sensors' support
I read your paper on Isaac Orbit. In the future work section, you mentioned "Further enhancements to the framework include integrating tactile sensors and 6-axis force-torque sensors."
I saw that Orbit supports contact sensors and my understanding is they measure the contact forces between rigid bodies. My questions is what is the difference between the already supported contact force sensors and the said tactile sensors? Are these tactile sensors specifically referring to Gelsight family sensors?

Look forward to your answer.

pascal-roth, 2024-04-08:
@Mayankm96 can you elaborate on that?

Mayankm96, 2024-04-26:
Hi @ruoshiwen ,

Yes, this corresponds to actual tactile sensors (such as Gelsight). Contact sensors only give you the net external normal contact force on a body. This isn't the same as what a tactile sensor or force-torque/haptic sensor gives.