isaac-sim/IsaacLab #325: container name "orbit" is already in use by "services.orbit-ros2": invalid compose project

swstbecrpgmail, 2024-03-26:
After I install docker and download the code of orbit, I run the following code "./docker/container.sh start", it reports 
"[INFO] Building the docker image and starting the container in the background...                                                
WARN[0000] /home/a123/orbit/docker/docker-compose.yaml: `version` is obsolete 
"services.orbit-base": container name "orbit" is already in use by "services.orbit-ros2": invalid compose project
"
[image]

jsmith-bdai, 2024-03-26:
> After I install docker and download the code of orbit, I run the following code "./docker/container.sh start", it reports
> "[INFO] Building the docker image and starting the container in the [...]

Thanks for posting your question - we just ran into this on our side and are debugging the issue. It seems to be related to different versions of docker compose supporting different features. Sorry for the inconvenience!

In the meantime, a quick fix to get you going is to rename the orbit-ros2 service's `container_name` within `docker-compose.yaml` to something other than `orbit`. A more permanent fix that works regardless of your docker compose version will be coming soon.

Let us know if this works for you!

swstbecrpgmail, 2024-03-27:
This indeed works for me. Thank you very much for your prompt reply and this open-sourced project! Expect your work to be more and more influential!

closed 2024-04-12 by Mayankm96