isaac-sim/IsaacLab #328: [Question] Cannot install rsl_rl

overrize, 2024-03-26:
When I use `./orbit.sh -e rsl_rl` in conda enviroment orbit,it show these 
`ERROR: Could not find a version that satisfies the requirement rsl-rl (unavailable) (from omni-isaac-orbit-tasks[rsl-rl]) (from versions: none)
ERROR: No matching distribution found for rsl-rl (unavailable)
`

pascal-roth, 2024-04-08:
We couldn't reproduce this error on our site, can you try to just run `./orbit.sh -e` or install `rsl-rl` manually by running `python -m pip install "rsl_rl@git+https://github.com/leggedrobotics/rsl_rl.git"`

Mayankm96, 2024-04-12:
If the issue persists, please reopen the issue with more information. Right now, we are unable to reproduce the issue on our end.

closed 2024-04-12 by Mayankm96

overrize, 2024-04-12:
It works after I recreate a new conda environment,this question is already solved,thanks