isaac-sim/IsaacLab #329: [Question] Optimal Code Directory for Spawning Assmeblies of Shapes

ErcBunny, 2024-03-26:
### Question

Hi,

I want to spawn abstracted robots from basic shapes, like the one demonstrated in the Isaac Sim Documentation [Working with USD](https://docs.omniverse.nvidia.com/isaacsim/latest/gui_tutorials/tutorial_intro_usd.html).

I see Orbit provides a Spawner package, but the package only allows for spawning simple and non-interlinked shapes (suboid, cylinder...), and from files.

I would like to have code that takes in geometry and physics params (e.g. wheel radius, body height, mass, inertia, collision bounding box), and spawns assemblies of the shapes (e.g. a cuboid representing the central part and two cylinders for the wheels), without needing to generate the USD file at first then use the "spawn from file" feature.

I wonder where is the optimal directory for implementing this?

Thanks.

Mayankm96, 2024-03-27:
Hi @ErcBunny, that is an interesting feature!

I think you can write your own spawning function that does this for you.

Regarding where to place it: It can be in `sim.spawners` directory with a new module named `assemblies.py` if it makes sense?

ErcBunny, 2024-03-27:
Hi @Mayankm96, thanks for the advice, I'll try to put it there!

ErcBunny, 2024-03-28:
So as I'm trying to add principal axes and moments of inertia attributes to the rigid body, I find that the function:

https://github.com/NVIDIA-Omniverse/orbit/blob/fcc216a5399661b6f3125e1e0a99235f073a3124/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/schemas/schemas.py#L302

lack support for these settings. Any suggestions on fixing this @Mayankm96 ? Thanks.