isaac-sim/IsaacLab #330: [Bug Report] Cannot run sensors tutorials code

gonzalezJohnas, 2024-03-27:
### Describe the bug

I cannot launch any of tutorials 04_sensors, the core dumped appear as soon as I use any sensors ( Camera, RayCasterCfg...). If I comment them everything works fine.

### Steps to reproduce

python source/standalone/tutorials/04_sensors/run_usd_camera.py

```
Program aborted due to an unhandled Error:
Unable to find target for this triple (no targets are registered)Aborted (core dumped)
```
[log.txt](https://github.com/NVIDIA-Omniverse/orbit/files/14772027/log.txt)

### System Info

Describe the characteristic of your environment:

- Commit: main branch
- Isaac Sim Version: 2023.1.1-rc.8+2023.1.688.573e0291.tc(orbit)
- OS: Ubuntu 20.04.6
- GPU: NVIDIA GeForce RTX 2080 Ti (sm_75)
- CUDA: 11.5
- GPU Driver: 525.85.05

### Additional context

So far all other tutorials works well, this error appear also in other tutorials when running in cpu mode. I am in remote server with no physical display connected to the desktop computer.
### Checklist

- [ x] I have checked that there is no similar issue in the repo (**required**)
- [ x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Can use the camera sensor for RL environment

Mayankm96, 2024-03-27:
I think this is because of the reason here:

> I am in remote server with no physical display connected to the desktop computer.

Does it work when you run headless?

gonzalezJohnas, 2024-03-27:
Everything works in headless mode, except for the sensor camera that's is why it is weird. I can successfully run other examples with this setup.

Mayankm96, 2024-03-27:
For running camera headless, you probably need to pass `--offscreen_render` flag as well.

gonzalezJohnas, 2024-03-27:
To clarify my problem, all other tutorial code works ( I can see the IsaacSim window open), just when getting to the sensor and more particularly run_usd_camera.py I have a core dumped.

gonzalezJohnas, 2024-03-27:
After debugging the program crash with the error:

>Program aborted due to an unhandled Error:
>Unable to find target for this triple (no targets are registered)Aborted (core dumped)

Only when using the CameraCfg object, other sensor works well (RayCaster, ContactSensor). The error appear either when calling **scene.reset()** or accessing **scene['camera']**.

gonzalezJohnas, 2024-04-02:
Hello, sorry to stress again about this bug but do you have some update on why the Camera sensors produce this error? Should I pull a specific version ??

YitianShi, 2024-04-04:
Hi I fixed somehow the same problem through:
```
sudo prime-select nvidia
```
if your 
```
sudo prime-select query
```
gives `on-demand`

gonzalezJohnas, 2024-04-08:
Thank you for the tip, but still have the error following your suggestion.

ninamortensen, 2024-05-23:
> Thank you for the tip, but still have the error following your suggestion.

Hi, I have the same issue, have you by any chance resolved this?

gonzalezJohnas, 2024-05-29:
For me plugging a monitor resolve the issue, I am still using ssh to work but that was the only solution to make the images works.